tock/tock #3229: Flash Controller <-> Capsule: Potential vulnerability.

twilfredo, 2022-09-19:
I've run into an interesting scenario while writing the initial version of [these tests](https://github.com/tock/tock/blob/master/boards/opentitan/src/tests/flash_ctrl.rs). Where in the initial commit it would erase and write over pages containing code. So during extended testing you may run into an `illegal_instruction` caused by the code that was overwritten. 

I'm wondering since the flash driver technically allows to R/W/Er all pages (at least for OpenTitan), and capsules that use the flash controller are 'untrusted' in Tock. They could potentially request a flash op on a page that should not be operated on. 
Does this fundamentally create point of vulnerability? 

OpenTitan `flash_ctrl` does allow setting memory protection regions for flash (similar to `pmp`). This could be implemented to mitigate the above, but then what about devices that don't have this functionality?

bradjc, 2022-09-20:
Hopefully I can capture how we think about this in Tock:

Capsules are untrusted, and therefore they should not be able to arbitrarily write flash, which could have security implications. However, capsules can (and should and do) have the ability to write flash, but that ability must be provided by trusted code. So, the trust lies in the board author verifying the capsule is not poorly implemented before passing a flash controller to the capsule. That trust is explicit in main.rs as the board must explicitly pass a reference to something that can write flash to the capsule.

What we have to ensure is that a capsule cannot write flash without that ability being explicit in main.rs. We ensure this because there is no way for a capsule to create a flash controller object without 1) being able to use `unsafe` and/or 2) having a particular chip crate as a cargo dependency.

alistair23, 2022-09-21:
> So, the trust lies in the board author verifying the capsule is not poorly implemented before passing a flash controller to the capsule

So in this case any capsule that has a `hil::flash::Flash` interface can read/write/erase any flash address. I thought that was against the [threat model](https://github.com/tock/tock/blob/master/doc/threat_model/Capsule_Isolation.md), but apparently it's not. It's just up to the board writer to verify the capsule will act correctly.

Still, considering that the OT hardware has access controls, it might be worth adding a new HIL that the board maintainer can use to enforce access in hardware (not sure how though, I haven't looked that closely at the hardware).

twilfredo, 2022-09-21:
> So in this case any capsule that has a `hil::flash::Flash` interface can read/write/erase any flash address. I thought that was against the [threat [...]

It would be good to implement, basically there are 8 memory protection regions, you can set each up with particular permissions and apply those permission to a specified range of pages (so 8 of these configs in total).  The granularity is pages though. 

Perhaps we could add HIL and then implement functionality in the flash driver for OT, things like removing erase/write over the code section and so on...

alistair23, 2022-09-21:
Considering we implement W^X on all OpenTitan memory via PMP I think it would be nice to do something similar via the flash controller

hudson-ayers, 2022-09-21:
> So in this case any capsule that has a `hil::flash::Flash` interface can read/write/erase any flash address. I thought that was against the [threat [...]

Technically an implementation of the `hil::flash::Flash` interface could restrict what addresses it allows access to, which would be one way to limit the damage that could be done by a capsule without requiring verification of the capsule itself -- the main issue is that currently there is no way to associate a particular client of the flash HIL with particular permissions, which means if you wanted to do something like "give TICKV access to only these 4 pages" that can only be done at the capsule level if you have multiple flash clients.

jrvanwhy, 2022-09-21:
> So in this case any capsule that has a `hil::flash::Flash` interface can read/write/erase any flash address. I thought that was against the [threat [...]

You're right about this being against the threat model -- one of the following should be true:

1. Capsules do not have access to a `hil::flash::Flash` implementation.
2. The `hil::flash::Flash` implementation limits what the capsule can write to.

This setup predates the written threat model, and should probably be revisited.

alistair23, 2022-09-22:
> Technically an implementation of the `hil::flash::Flash` interface could restrict what addresses it allows access to, which would be one way to limit the damage that could be done by a capsule [...]

Good point. So a solution would require keeping track of clients and also assigning permissions.

At least at a start saying that "this region of kernel code is write protected" would be something (although it would block the kernel from updating itself, but I think that's a fair trade off)

> This setup predates the written threat model, and should probably be revisited.

Agreed, unfortunately getting any changes to the flash HIL is a herculean task. Exposing OT's protections via an API would be a good start and help alleviate this a bit