tock/tock #3244: RFC: Uunsafe `WriteOnly`, `ReadWrite`, and `ReadOnly` (?) variants for tock-register-interface?

andre-richter, 2022-09-26:
In application grade processors, there potentially are registers that can compromise Rust's memory safety one way or the other. For example, registers that control virtual memory could be used to install page tables that create aliases which would break Rust's memory safety guarantees.

Manipulating such registers should be guarded with `unsafe` to stay true to Rust. If such registers should be implemented using tock-register-interface, unsafe variants of the `WriteOnly`, `ReadWrite`, and maybe `ReadOnly` interfaces (couldn't think of a real-world example for a register yet that compromises memory safety by just reading it) would be needed.

I am not sure if `tock` itself currently supports processor architectures that could make use of this. [`cortex-a`](https://github.com/rust-embedded/cortex-a), though, which is one of the more prominent users of tock-register-interface, would definitely embrace it.

So I wanted to first check with you if there is a general willingness to add unsafe interfaces variants?

jrvanwhy, 2022-09-26:
> So I wanted to first check with you if there is a general willingness to add unsafe interfaces variants?

I am supportive of adding unsafe interface variants.

I have been working on a rewrite of the `register_structs!` macro and register types that would collide with work on adding unsafe interface variants. My new design makes it easier to add and compose these variants (e.g. you could specify that one register is safe to read and unsafe to write, while another is unsafe to read or write), so it's probably worth waiting for the redesign to add this functionality.

andre-richter, 2022-09-26:
That sounds great! We can wait, no urgency here. It's just something that crossed my mind recently and that I'd like `cortex-a` to do better in the future than it does currently.

ppannuto, 2022-09-27:
+1 here that we would be happy to add support for something like this.

The only potential caveat I can think of is that we are always a bit wary of code size. However, my expectation is that an implementation of this would likely end up looking something like a sum of traits, and if Tock platforms don't use any of the unsafe traits, we won't see any size impact.

lschuermann, 2022-09-28:
We've had discussions like this before and previously sided with a different line of argument: https://github.com/tock/tock/issues/2593#issuecomment-847139818

However, that was mostly concerned with register accesses which are conditionally valid, and may otherwise invoke undefined behavior. I don't know whether it makes sense to make a distinction between registers which are only conditionally safe to access and where the access itself may invoke potentially undefined behavior (e.g. invoke the trap handler, reset the chip, etc.), from registers which are inherently safe to access but can violate expectations of Rust's machine model (e.g. not mapping a mutable address space section twice, etc.). In fact I would argue that an incorrect MPU configuration, for instance revoking read permissions required by the kernel, is equally bad, as none of this can be represented in Rust's execution model. Other current examples include DMA peripherals, flash controllers, etc.

I am not entirely convinced that we need to diverge from our previous stance on this, which is to put the burden of ensuring safety onto the developer when a `tock-registers` struct is instantiated. Nonetheless, I do see the need to at least improve our documentation in this regard, and check the existing code base to ensure that registers generally are not accessible outside of the intended scope (i.e. the respective peripheral driver).

andre-richter, 2022-09-28:
Now that you wrote it, I remember reading along (and sometimes even participating in…) these discussions in the past.
It’s been some time and my memory had failed me.

The ones I remember now were about slightly different, easier, examples, not exactly the memory (re-)mapping example I brought up here, which feels like a very special case.

I’ll close this for now and ponder a bit where to go with the cortex-a case.

Thanks everyone for your input.

closed 2022-09-28 by andre-richter

ppannuto, 2022-09-28:
> not exactly the memory (re-)mapping example I brought up here, which feels like a very special case.

Indeed, it was this example that convinced me this made sense, as I understand that to create actual unsoundness in the Rust context. Happy to defer this discussion to specific examples in the future, however.

jrvanwhy, 2022-09-28:
"Should all register operations in tock-registers be unsafe?" and "Should tock-registers support unsafe register operations?" are very different questions.

Making all register operations `unsafe` destroys the utility of `tock-registers`, as it would make `tock-registers` no safer or more ergonomic than `core::ptr::{read_volatile,write_volatile}`. Making a few registers operations `unsafe` in the middle of an otherwise safe register struct keeps most of `tock-registers`' safety and utility, while allowing it to soundly be used with more peripherals.

kupiakos, 2024-05-31:
> which is to put the burden of ensuring safety onto the developer when a tock-registers struct is instantiated

This is asking the user to verify all safe usage of a type after its unsafe construction, a task that becomes more difficult the larger the code scales, especially as it is edited. This is similar to `Pin::get_unchecked_mut`:  it's sound to define an API that does this, but an unsafe interface should always try to minimize the code that must be checked for correctness. It removes the ability to have eager `unsafe` checks inside of the driver that could actually exercise the UB, making it much less resilient to edits. Imagine a reviewer: "this diff doesn't touch any `unsafe` code, so it's won't affect soundness".