tock/tock #3247: tbf parsing dramatically increases stack usage

ppannuto, 2022-09-27:
I ended up tracking this down from @gemarcano's testing in #3240. The hifive1 board was crashing due to stack overflows. Digging into it a bit, it looks like https://github.com/tock/tock/pull/2824 tripled stack usage (!). This isn't something we necessarily actively monitor per se, though maybe we should consider adding `make stack-analysis` to the benchmarks workflow. 

From a bisect:

f3f3f5e    tock-tbf: Parse the Driver Permissions
grows the stack from 416 to 944 bytes

and then the next one in the series:
558ff61    tock-tbf: Parse the Persistent ACLs
grows the stack from 944 to 1200, which is basically what it is today

I don't think there's anything fundamental there that should need so much stack, so at some point it would probably be good to look into a refactor to bring this back down.

bradjc, 2022-09-28:
Is the issue that the stack is being used in main() and not returned? Parsing TBF headers should only happen when processes are loaded. Or, does that stack space need to be preserved for the life of the kernel?

ppannuto, 2022-09-28:
Without having dug into the code to confirm this yet, my hunch is that (1) the current implementation is using stack and not returning it, and (2) that this is not actually necessary.

alistair23, 2022-09-28:
That's a pain! I'll have a look and report back

gemarcano, 2022-09-29:
I've been chatting on the Tock slack about this for a bit while I was trying to investigate what was the source of the increase, but I think I reached the limits of my ability to debug Tock/rust. It was suggested to me to try to `#[inline(never)]` `load_processes`, as that's where the TBF loading code is, but that actually increased stack usage. I tried determining what's on the stack and what isn't, but as I don't completely understand the rust memory model yet, and all of the inlining that's being done, I'm currently unable to continue.

One final experiment I want to run is to sprinkle some watchpoitns throughout the main stack frame to see where it accessed, and from there back out some of the structures that are in that stack frame.

gemarcano, 2022-09-29:
I started seeing `peripherals` being accessed from that stack allocated memory, and I was wondering why that was the case as I thought that should have been in static memory... then I remembered what I had read previously from #2425, and apparently the HiFive1 did not have that fix applied. With making `get_peripherals` its own function, I get a massive reduction in the main stack frame size, from 1200 to 112.

However, `load_processes` is still reporting a stack frame larger than 1000, and that's likely due to TBF parsing.

gemarcano, 2022-09-29:
Sorry, I was a bit hasty. This large reduction in stack frame from main was only possible by implementing `get_peripherals` _and_ marking `load_processes` as `#[inline(never)]`. I don't see any improvements with just one of those two changes.

alistair23, 2022-09-29:
Ok. I had a look at this. For the OpenTitan board:

Before f3f3f5e0a2f3d75839c18a4fc145530f01530793

```
earlgrey-cw310
----------------------
   main stack frame: 
           32     main

   5 largest stack frames: 
          640     _ZN6kernel6kernel6Kernel11kernel_loop17h1523970983234a3bE
          592     _ZN14earlgrey_cw3105setup17h4c6f81c8b785225fE
          400     _ZN95_$LT$kernel..process_standard..ProcessStandard$LT$C$GT$$u20$as$u20$kernel..process..Process$GT$18print_full_process17h20716e54c1b81247E
          384     _ZN95_$LT$kernel..process_standard..ProcessStandard$LT$C$GT$$u20$as$u20$kernel..process..Process$GT$11try_restart17hb9d3d64b3b03970dE
          304     _ZN6kernel5debug11panic_print17hbedb3649dbc762a4E

```

After f3f3f5e0a2f3d75839c18a4fc145530f01530793

```
earlgrey-cw310
----------------------
   main stack frame: 
           32     main

   5 largest stack frames: 
         1088     _ZN14earlgrey_cw3105setup17h4c6f81c8b785225fE
          640     _ZN6kernel6kernel6Kernel11kernel_loop17h1523970983234a3bE
          400     _ZN95_$LT$kernel..process_standard..ProcessStandard$LT$C$GT$$u20$as$u20$kernel..process..Process$GT$18print_full_process17h20716e54c1b81247E
          384     _ZN95_$LT$kernel..process_standard..ProcessStandard$LT$C$GT$$u20$as$u20$kernel..process..Process$GT$11try_restart17hb9d3d64b3b03970dE
          304     _ZN6kernel5debug11panic_print17hbedb3649dbc762a4E

```

So setup() went from `592` bytes of stack usage to `1088` bytes!

Where does that come from?
 * Adding `Option<TbfHeaderV2Permissions<8>>` to `types::TbfHeaderV2` adds `176` bytes
     * `TbfHeaderV2Permissions<8>` is `130` bytes worth of data being passed via the stack, not sure where the other bytes come from
 * The `try_from()` implementation of `TbfHeaderV2Permissions` adds `320` bytes, which almost all comes from returning `perms`

I'm not sure how we can improve this though. The problem is that the TBF data is passed around on the stack and `TbfHeaderV2Permissions` is a large-ish struct.

I've tried converting `TbfHeaderDriverPermission::allowed_commands` to a `u32` from a `u64`, but that only saves `140` bytes.

Converting `TbfHeaderV2Permissions<8>` to `TbfHeaderV2Permissions<4>` saves `172` bytes, but that is a little limiting.

Both options above saves `284` bytes.

Any other ideas?

hudson-ayers, 2022-10-04:
I don't really think we should be worrying about stack frames less than 2kB which are popped off the stack before the kernel loop runs. If there is a problem it is caught immediately at boot. And we already know that the hifive1 has an unacceptably low amount of RAM to be a good target for Tock -- I don't think we should make compromises for other boards to accommodate it. That said, making `allowed_commands` a `u32` might be fine, and a way to gain a little savings here.

gemarcano, 2022-10-04:
I think managing the stack frames in `main` is also the better approach, as it's something that would benefit all boards (hopefully with minimal to no cost). From my own testing, however, it's rather difficult currently to tell what needs to be inlined such that rustc/llvm doesn't inline a lot of the stack allocations. Specific to this issue, I had to implement two changes so that the `load_processes` stack frame (where the TBF functions are called) wasn't just inlined in `main`'s stack frame:
 1. set `load_processes` as `#[inline(never)]`
 2. make a `get_peripherals` function for the HiFive1 board (also as `#[inline(never)]`) so that the stack frame from the peripheral static initialization is also released.

Without _both_ of these modifications I was still seeing a "large" stack frame of over 1000 bytes, but with both, the `main` stack frame size reported by `make stack-analysis` dropped to around 100 bytes! I'm not a huge fan of me having to hack around in the kernel crate, though.

hudson-ayers, 2022-10-04:
I think we should definitely use get_peripherals for the hifive1, it is a bug that it doesnt have it when all the other boards do.

the problem with `#[inline(never)]` for `load_processes()` in the kernel crate is that inlining it is actually beneficial for code size and runtime speed on other boards, which don't have issues with memory space but do have issues with flash space. One option to fix this for just the hifive1 without hacking around in the kernel crate would just be to define a wrapper function for load_processes in main.rs for the hifive1, e.g.

```rust
#[inline(never)]
fn load_processes_not_inlined(board_kernel: &'static Kernel, chip: &'static Chip) {
    kernel::process::load_processes(
        board_kernel,
        chip,
        core::slice::from_raw_parts(
            &_sapps as *const u8,
            &_eapps as *const u8 as usize - &_sapps as *const u8 as usize,
        ),
        core::slice::from_raw_parts_mut(
            &mut _sappmem as *mut u8,
            &_eappmem as *const u8 as usize - &_sappmem as *const u8 as usize,
        ),
        &mut PROCESSES,
        &FAULT_RESPONSE,
        &process_mgmt_cap,
    )
    .unwrap_or_else(|err| {
        debug!("Error loading processes!");
        debug!("{:?}", err);
    });
}
```

gemarcano, 2022-10-04:
Weirdly enough, the `get_peripheral` approach alone does nothing to the stack utilization (before and after are both reporting 1200 bytes of utilization through `make stack-analysis`). I had to move a couple of more things around to get the `load_processes_not_inlined` to compile including making it unsafe (needs exports from linker), but applying both `get_peripheral` and this `no_inline` wrapper approach brings the `main`  stack utilization to 96!

I think I'll PR these changes for the HiFive1, since they only affect it and the tradeoff in additional flash memory required (832 growth in kernel binary size) is acceptable.

hudson-ayers, 2022-10-04:
> Weirdly enough, the `get_peripheral` approach alone does nothing to the stack utilization (before and after are both reporting 1200 bytes of utilization through `make stack-analysis`). I had to move [...]

I suspect that with both not marked as no_inline, one of two things is happening:
Either 1) LLVM is smart enough to realize both of those stack uses are temporary and reuses the same stack space for each or
2) LLVM chooses to not inline the peripheral code anyway because the presence of the other inlined function changes the inlining heuristic cost function enough

bradjc, 2022-10-12:
We can close this?

hudson-ayers, 2022-10-12:
Addressed by https://github.com/tock/tock/pull/3265

closed 2022-10-12 by hudson-ayers