tock/tock #3257: New register blocks macro doesn't support conditional compilation

berkus, 2022-10-01:
Hi!

In the old tock-registers I did the following:

```
        (0x90 => __reserved_11),
        #[cfg(feature = "rpi3")]
        (0x94 => pub PUD: ReadWrite<u32>), // pull up down
        #[cfg(feature = "rpi3")]
        (0x98 => pub PUDCLK: [ReadWrite<u32>; 2]),
        #[cfg(feature = "rpi3")]
        (0xa0 => __reserved_12),
        #[cfg(feature = "rpi4")]
        (0xe4 => PullUpDownControl: [ReadWrite<u32>; 4]),
        (0xf4 => @END),
```

In the new version, it throws an error 

```
    --> /Users/berkus/.rustup/toolchains/nightly-aarch64-apple-darwin/lib/rustlib/src/rust/library/core/src/panic.rs:54:9
     |
48   |   pub macro panic_2021 {
     |   -------------------- in this expansion of `$crate::panic::panic_2021!` (#7)
...
54   |           $crate::panicking::panic_display(&$arg)
     |           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ the evaluated program panicked at 'Invalid end offset for field PUD (expected 228 but actual value differs)', machine/src/platform/rpi3/gpio.rs:42:1
     |
```

I've traced this down to the conditional compilation flags; but I'm wondering was it broken somehow before (although gpio code worked on rpi4 using the PullUpDownControl register just fine) or this is some recent regression?

lschuermann, 2022-10-04:
Thanks for bringing this to our attention.

The panic during compilation is desired if certain sanity checks in `tock-registers` happen to fail. Previously these were implemented as automatically generated tests, which presumably would've also failed on the example provided. Hence I am not sure if this should be classified as a regression.

Nonetheless, simply "removing" fields by means of tagging them with a `#[cfg(...)]` is not a supported mode of use for `register_structs!`, as it needs to validate that the generated struct layout matches the requested addresses at which fields shall be located. Take the following register definition:
```rust
register_structs! {
    MWE {
	(0x0 => a: ReadWrite<u32>),
	(0x4 => b: ReadOnly<u32>),
	(0x8 => c: WriteOnly<u32>),
	(0xc => @END),
    }
}
```

This will expand to:
```rust
#[repr(C)]
struct MWE {
    a: ReadWrite<u32>,
    b: ReadOnly<u32>,
    c: WriteOnly<u32>,
}
```

However, when removing `(0x4 => b: ReadOnly<u32>),` from the definition and disabling compile-time checks, the generated struct clearly violates the desired layout constraints:
```rust
#[repr(C)]
struct MWE {
    a: ReadWrite<u32>,
    c: WriteOnly<u32>,
}
```

We can fix this by unconditionally adding a `_reserved` field to the struct, which will automatically insert appropriate padding to align with the next field's address:
```rust
register_structs! {
    MWE {
	(0x0 => a: ReadWrite<u32>),
	(0x4 => _reserved0),
	(0x4 => b: ReadOnly<u32>), // can now be safely removed
	(0x8 => c: WriteOnly<u32>),
	(0xc => @END),
    }
}
```
generates
```rust
#[repr(C)]
struct MWE {
    a: ReadWrite<u32>,
    _reserved0: [u8; 0x4 - 0x4],
    b: ReadOnly<u32>,
    c: WriteOnly<u32>,
}
```
or, when removing field `b`:
```rust
#[repr(C)]
struct MWE {
    a: ReadWrite<u32>,
    _reserved0: [u8; 0x8 - 0x4],
    c: WriteOnly<u32>,
}
```

berkus, 2022-10-11:
@lschuermann as can be seen in my code, the padding fields remain to be present even with conditional compilation triggering a different layout 

```
       (0x90 => __reserved_11),
        #[cfg(feature = "rpi3")]
        (0x94 => pub PUD: ReadWrite<u32>), // pull up down
        #[cfg(feature = "rpi3")]
        (0x98 => pub PUDCLK: [ReadWrite<u32>; 2]),
        #[cfg(feature = "rpi3")]
        (0xa0 => __reserved_12),
        #[cfg(feature = "rpi4")]
        (0xe4 => PullUpDownControl: [ReadWrite<u32>; 4]),
        (0xf4 => @END),
```

Has three transforms:

1. No rpi4 or rpi3 defined

```
        (0x90 => __reserved_11),
        (0xf4 => @END),
```

2. rpi3 defined

```
       (0x90 => __reserved_11),
        (0x94 => pub PUD: ReadWrite<u32>), // pull up down
        (0x98 => pub PUDCLK: [ReadWrite<u32>; 2]),
        (0xa0 => __reserved_12),
        (0xf4 => @END),
```

3. rpi4 defined

```
        (0x90 => __reserved_11),
        (0xe4 => PullUpDownControl: [ReadWrite<u32>; 4]),
        (0xf4 => @END),
```

As you can see, in all cases layouts are valid.

ppannuto, 2022-10-12:
This is interesting – hw that is supersets of functionality is not uncommon (indeed, there are m0/m3/m4 core registers that follow this pattern as well IIRC).

This seems like something the registers library can/should help support more explicitly perhaps. I might imagine some kind of "only on `target_arch=xxx`" (though need platform, which may be less standard :/, it would be nice to use formal compiler metadata for this rather than arbitrary features if we can...). This could resolve marked symbols either to their actual value, or an autogenerated `reserved` identifier based on the platform being compiled.

jrvanwhy, 2022-10-12:
I think it is really hard to support conditional register compilation while also supporting unit-testing functionality (support for fake or mock peripherals).

phil-levis, 2022-10-14:
I think this is exactly the kind of configuration conditional compilation we want to avoid; when there is just one `cfg` it's one thing, but for complex families this can be come an intricate expression. Register definitions don't change, so the standard approach is to explicitly write out the definitions for each chip. E.g., even Contiki does it:

https://github.com/contiki-os/contiki/blob/master/cpu/arm/at91sam7s/AT91SAM7S64.h

As well as FreeRTOS:

https://github.com/ARM-software/CMSIS-FreeRTOS/blob/main/Demo/ARM7_STR75x_GCC/STLibrary/inc/75x_uart.h

This also adds isolation between hardware versions: someone fixing the register definitions for hardware X doesn't accidentally break them for hardware Y (and they don't have Y to test).

RIOT does conditional includes and I'm terrified:

https://github.com/RIOT-OS/RIOT/blob/master/cpu/nrf52/include/cpu_conf.h

berkus, 2022-10-15:
> just one `cfg` it's one thing, but for complex families this can be come an intricate expression. Register definitions don't change, so the standard approach is to explicitly write out the definitions for each chip.
> This also adds isolation between hardware versions: someone fixing the register definitions for hardware X doesn't accidentally break them for hardware Y (and they don't have Y to test).

Totally agree with this point and fixed, currently, by spelling two copies of the same register definitions for now.  But oh boy are they the same, I just don't like such wet code. RPi3 and RPi4 are basically (not everywhere but close) the same thing and whenever possible I'd like to keep them close in definitions.