tock/tock #3266: linkage error when compiling esp32-c3-devkitM-1 kernel

CosminGGeorgescu, 2022-10-05:
When trying to run `make` for the esp32-c3-devkitM-1 board i get the following errors with the rust-lld linker:

```
   Compiling esp32-c3-board v0.1.0 (/home/pas/project/tock/boards/esp32-c3-devkitM-1)
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/tmp/rustcj8pLkK/symbols.o" "/home/pas/project/tock/target/riscv32imc-unknown-none-elf/release/deps/esp32_c3_board-fec9d415f72a6ba8.esp32_c3_board.3818b060-cgu.0.rcgu.o" "--as-needed" "-L" "/home/pas/project/tock/target/riscv32imc-unknown-none-elf/release/deps" "-L" "/home/pas/project/tock/target/release/deps" "-L" "/home/pas/.rustup/toolchains/nightly-2022-07-03-x86_64-unknown-linux-gnu/lib/rustlib/riscv32imc-unknown-none-elf/lib" "-Bstatic" "/home/pas/project/tock/target/riscv32imc-unknown-none-elf/release/deps/libcompiler_builtins-50f34a461c299cbb.rlib" "-Bdynamic" "-znoexecstack" "-L" "/home/pas/.rustup/toolchains/nightly-2022-07-03-x86_64-unknown-linux-gnu/lib/rustlib/riscv32imc-unknown-none-elf/lib" "-o" "/home/pas/project/tock/target/riscv32imc-unknown-none-elf/release/deps/esp32_c3_board-fec9d415f72a6ba8" "--gc-sections" "-L/home/pas/project/tock/boards/esp32-c3-devkitM-1" "-Tlayout.ld" "-nmagic" "-icf=all"
  = note: rust-lld: error: undefined symbol: __atomic_load_4
          >>> referenced by deferred_call.rs:33 (kernel/src/deferred_call.rs:33)
          >>>               /home/pas/project/tock/target/riscv32imc-unknown-none-elf/release/deps/esp32_c3_board-fec9d415f72a6ba8.esp32_c3_board.3818b060-cgu.0.rcgu.o:(_$LT$esp32..rng..Rng$u20$as$u20$kernel..hil..entropy..Entropy32$GT$::get::h8efc3363f532753f)
          
          rust-lld: error: undefined symbol: __atomic_store_4
          >>> referenced by deferred_call.rs:41 (kernel/src/deferred_call.rs:41)
          >>>               /home/pas/project/tock/target/riscv32imc-unknown-none-elf/release/deps/esp32_c3_board-fec9d415f72a6ba8.esp32_c3_board.3818b060-cgu.0.rcgu.o:(_$LT$esp32..rng..Rng$u20$as$u20$kernel..hil..entropy..Entropy32$GT$::get::h8efc3363f532753f)
          

error: could not compile `esp32-c3-board` due to previous error
make: *** [../Makefile.common:423: /home/pas/project/tock/target/riscv32imc-unknown-none-elf/release/esp32-c3-board] Error 101
```

here is a link to my project branch: 
[](https://github.com/CosminGGeorgescu/tock/tree/esp32-c3-devkitM-1_rng)

the error arises from **line #59** in the **/chips/esp32/src/rng.rs** when setting a deferred_call:
[](https://github.com/CosminGGeorgescu/tock/blob/535d8131bf1863a3754963aa276b53b40a574044/chips/esp32/src/rng.rs#L59)

hudson-ayers, 2022-10-05:
Ah...I was wondering when we would hit this. It seems you are the first person to try to actually use chip-level deferred calls on a `riscv32imc` platform.

My understanding is that atomics support via `core_intrinsics` on riscv32imc platforms is effectively currently broken even for simple loads/stores because of changes in LLVM. The discussion around this is complex and very difficult to follow: https://github.com/rust-lang/rust/issues/99668 might give you a taste. I tried to figure out whether https://reviews.llvm.org/D130621 would fix this problem once Rust picks up LLVM16, but could not get to a point where I was confident in the answer.

It is also possible that https://github.com/rust-lang/rust/pull/98333 is the fix we should be waiting on.

Alternatively, if we do not want to wait on Rust/LLVM to fix this for us, we can go forward with removing our reliance on atomic types at all in Tock via something like https://github.com/tock/tock/pull/3086

alexandruradovici, 2022-10-06:
@hudson-ayers for the time being it might seems easier to move forward with #3086 as this seems to be something that we can do, rather than wait for Rust or LLVM.

hudson-ayers, 2022-10-10:
#3086 commits us a little more to a particular `DeferredCall` approach which is incompatible with a couple other approaches I would like to make work but that do rely on atomics (but haven't had time to fully explore). so I don't think it is something that will necessarily be merged in the short term either. You may want to just use `DynamicDeferredCall` instead, for now, despite the unfortunate size hit.

bradjc, 2023-08-11:
Fixed in #3382 and #3574.

closed 2023-08-11 by bradjc