tock/tock #3270: Rename get_peripherals in boards' initialization to create_peripherals

gemarcano, 2022-10-06:
In PR #3265 it was suggested to rename the `get_peripherals` function (idea was introduced into the code by PR #2503 I think) to `create_peripherals`, as it really can't (shouldn't?) be called multiple times. Many boards use the `get_peripherals` function name. Should these all be renamed as well?

Additionally, as these functions should only ever be called once, should there be some guard to ensure they're called only once?

hudson-ayers, 2022-10-06:
They will panic if called twice, so no need for a guard, but I would approve a PR renaming them all

gemarcano, 2022-10-07:
One benefit of refactoring the names would be consistency. Any new developers porting a new board will likely adopt the most common idiom in the current tree (as I did by looking at how other boards did it), which is currently `get_peripherals`.

bradjc, 2022-10-07:
Don't we have consistency now?

```
❯ git grep get_peripherals | wc -l
      46
❯ git grep create_peripherals | wc -l
       0
```

gemarcano, 2022-10-07:
We do, but I was asked to rename my implementation of `get_peripherals` to `create_peripherals` for #3265 . So what should I do in that commit then? Should we stick to `get_peripherals`?

alistair23, 2022-10-09:
`get_peripherals()` sounds like you can call it multiple times to get a pointer to a struct

While `create_peripherals()` is more accurate as the function is creating the struct.

As Hudson says, we enforce that `get_peripherals()` can only be called once, but it's a confusing and misleading name. It does seem worth changing them all