tock/tock #3278: Subscribe error code for a nonexistent driver_num

jrvanwhy, 2022-10-07:
[TRD 104](https://github.com/tock/tock/blob/master/doc/reference/trd104-syscalls.md#421-the-null-upcall) says:

If a userspace process invokes subscribe on a driver ID that is not installed in the kernel, the kernel MUST return a failure with an error code of `NODEVICE` and an upcall of the Null Upcall.

However, while implementing `libtock-rs`, I [observed](https://github.com/tock/libtock-rs/blob/1d785a043a95d83b410f6a099a6121fc101ca3b7/unittest/src/fake/syscalls/subscribe_impl.rs#L76) that as of Tock 2.0, the kernel returns `NOMEM` instead of `NODEVICE`.

I do not know what Tock 2.1 does -- Subscribe's implementation was overhauled between 2.0 and 2.1 -- but at least in 2.0 we had an implementation/documentation mismatch.

alexandruradovici, 2022-10-10:
#3276 fixes this issue.

alexandruradovici, 2022-10-10:
As far I noticed, 2.1 does not return `NODEVICE` for any of the `subscribe` and `allow` system calls.

jrvanwhy, 2022-10-13:
Best as I can tell, here are the documented and actual return codes for stable Tock releases:

| Syscall                  | TRD 104    | Tock 2.0     | Tock 2.1   |
| ------------------------ | ---------- | ------------ | ---------- |
| Command                  | `NODEVICE` | `NODEVICE`   | `NODEVICE` |
| Read-Only Allow          | `NODEVICE` | `NODEVICE`   | `NOMEM`    |
| Read-Write Allow         | `NODEVICE` | `NODEVICE`   | `NOMEM`    |
| Subscribe                | `NODEVICE` | `NOMEM`      | `NOMEM`    |
| Userspace-Readable Allow | `NODEVICE` | Didn't exist | `NODEVICE` |

alexandruradovici, 2022-10-14:
IMHO I think we would treat this as a bug. Based on @jrvanwhy table, I think it is clearly visible that this has been an honest implementation mistake.

The arguments being:
1. this was not an intentional change, we simply do not have any unit tests (I guess we should) and nobody actually checked that the new `subscribe` in 2.0 and new `allow` in 2.1 where actually following TRD 104
2. semantically they should return `NODEVICE` as this is the actual error. A user will not be able do distinguish between a driver that does not exist and a driver that does not properly allocate a grant.
3. Tock 2.1 was recently release, I would suggest releasing 2.1.1 with this bug fix. I think it is reasonable to believe that not many projects actually switched to the new version (or if they did, they will have an issue with `allow` that does return different from 2.0)
4. I don't think libtock-c is impacted in any way, as `NODEVICE` does not seem to be used in any place (https://github.com/tock/libtock-c/search?q=NODEVICE).
5. libtock-rs does not have any stable release yet so we should be able to change it before releasing for 2.1

jrvanwhy, 2022-10-14:
This was discussed on the Tock core WG, and the conclusion was to change the return code to `NODEVICE` for all syscalls and to release 2.1.1 with the change. The change PR should be limited to the fix, and not part of a larger refactoring PR (i.e. we shouldn't merge all of #3276 into 2.1.1).

lschuermann, 2022-11-01:
With #3276 merged into master and #3300 into `release-2.1`, I believe the only thing missing here is a tag for `v2.1.1`. I'll create a PR to update the changelog on the `release-2.1` issue and tag a release afterwards to close this issue.