tock/tock #3311: tock-registers: CLEAR FieldValue does not perform match check with inverted mask

Fighter19, 2022-11-02:
Consider following test case:
```rust
        #[test]
        fn test_clear_match() {
            register_bitfields! {
                u32,
            
                TEST [
                    FLAG OFFSET(18) NUMBITS(1) [],
                ]
            }

            let value = 266241;
            match TEST::FLAG::CLEAR.matches_any(value) {
                true => panic!("Field Value should not have matched {}", TEST::FLAG::CLEAR.mask()),
                false => (),
            }
        }
```

Even though the test value has bit 18 set, the result of the function returns true.

brghena, 2022-11-02:
266241 -> 0x41001 which sets bits 0, 12, and 18. So I agree that this code _should_ definitely work.

Do you know if it's a problem with all functions, or just `matches_any()` maybe?

Fighter19, 2022-11-02:
I only noticed it with this particular function.

hudson-ayers, 2022-11-26:
I see two issues here:

**Poor Name**

`matches_any()` is a poor name for the functionality it implements, it would be better named `is_any_set()`, since it just checks if any bits in the passed parts of the field are set -- hence why the provided example using `TEST::FLAG::CLEAR` does not work. It would also work incorrectly for a multi-bit field:
```rust
//! dma.rs

register_bitfields![u32,
    Mode [
        /// Size of Transfer
        SIZE OFFSET(0) NUMBITS(2) [
            Byte = 0,
            Halfword = 1,
            Word = 2
        ]
    ],
];

// The below variable will be `true` if the dma register is set to either `Halfword` 
// or `Word`, which is clearly incorrect
let is_halfword_transfer = dma_regs.matches_any(Mode::SIZE::HALFWORD);
```

I don't think it is possible to provide a "correct" implementation for the current interface, since the current interface relies on multiple `FieldValue`s being added into a single `FieldValue`, but a correct implementation which could actually check on a field-by-field basis for any matches would need to retain the information about which individual fields were passed.

**I think we should rename the current method to `any_bits_set()`. If we want to be able to provide the functionality suggested by the current name, we would need a new interface:**

```rust
trait Readable {
    #[inline]
    /// Check if any specified parts of a field match
    fn matches_any(&self, fields: &[FieldValue<Self::T, Self::R>]) -> bool {
        for field in fields {
            if self.get() & field.mask() == field.value {
                return true;
            }
        }
        return false;
    }
}
```

I think there is also a second, semi-related issue with this function:

**Unclear Argument Order** 
There are two implementations of `matches_any()` on different structs -- the implementation on the `Readable` trait, and the implementation on the `FieldValue` struct:
```rust
trait Readable {
    #[inline]
    /// Check if any specified parts of a field match
    fn matches_any(&self, field: FieldValue<Self::T, Self::R>) -> bool {
        field.matches_any(self.get())
    }
}

impl<T: UIntLike, R: RegisterLongName> FieldValue<T, R> {
    /// Check if any specified parts of a field match
    #[inline]
    pub fn matches_any(&self, val: T) -> bool {
        val & self.mask != T::zero()
    }
}
```

Notably, the first implementation effectively reverses the order of arguments when calling the second. Also notably, both implementations are public. This on its own is a little confusing -- the canonical example for `matches_any()` is only provided for the `Readable` method:

```rust
            let any_interrupts = registers.s.matches_any(Status::TXINTERRUPT + Status::RXINTERRUPT);
```

And does not match how you would want to use it to check if a given value has any bits set for a given field, which looks more like the code from this issue:
```rust
Status::TXINTERRUPT.matches_any(value_taken_from_register);
```

I think we might want to consider making the `FieldValue` implementation of the method `pub(crate)`, since it is hard to imagine a good reason to use it rather than calling it on whatever register you want to check for a match. Any use cases covered by the currently public implementation could be covered by creating a `LocalRegisterCopy` initialized to the value in question.

closed 2023-03-08 by bors[bot]