tock/tock #3319: FlashController HIL offset argument usage

teonaseverin, 2022-11-07:
We are trying to implement the FlashController HIL for a flash driver on a `stm32f429` device and we were wondering what the role of the `offset` argument in the `FlashController::read_region` function is. Is it possible to read data from anywhere in the memory and not just the start of a region/sector?  

@alistair, I saw that you implemented most of the tests. I was curious if, for example the implementation at line 434 in the file `libraries/tickv/src/[tests.rs](http://tests.rs/` doesn’t overflow, if the `offset` argument is different than 0.
```rust
fn read_region(
    &self,
    region_number: usize,
    offset: usize,
    buf: &mut [u8; 256],
) -> Result<(), ErrorCode> {
    println!("Read from region: {}", region_number);

    for (i, b) in buf.iter_mut().enumerate() {
        *b = self.buf.borrow()[region_number][offset + i]
    }

    Ok(())
}
```

Maybe it’s something we misunderstood.

alistair23, 2022-11-08:
There seems to be some confusion, so first I wanted to clarify something.

A HIL in Tock is a [Hardware Interface Layers](https://github.com/tock/tock/blob/master/doc/reference/trd2-hil-design.md). It's defined in the kernel and typically used to expose drivers to capsules.

In this case you are talking about the `FlashController` trait as part of the TicKV key-value store. If you are using Tock, you don't need to implement `FlashController` as it's implemented for you as part of the `TickFSFlastCtrl` in the `tickv` capsule.

So I'm assuming you are trying to use the tickv library in your own project. In which case you will need to implement `FlashController` for your hardware.

The idea of the `offset` field in `read_region()` is to allow reading an offset into a flash region. Flash regions can be very large so the idea is to allow reading only part of the region. In some hardware you can do this directly, while in other hardware the driver might need to perform a full read before just dropping the first `offset` bytes.

In saying that the size of `buf` is fixed at length `S`, so you still need to supply a full length buffer, it just won't be fully populated.

Looking at the implementation again though, we never supply a non-zero `offset`. So although it's an argument to the function, it's currently always zero. So you could easily just report an error if the value is non zero in your implementation.

I can't currently think of a use case where we would supply an non-zero value, so it might be worth removing the `offset` argument.

teonaseverin, 2022-11-09:
Thank you for your answer.

I just realised my mistake, I was indeed talking about the FlashController trait.

So, if the `offset` field is used to read at a certain offset into a flash region, how much are we allowed to read? Until the buffer that is received as an argument is entirely filled or until the page boundary?

Also, I took a look at the `TickFSFlastCtrl` implementation in the `tickv` capsule and I don’t understand the meaning and usage of the `2048` constant. Is it supposed to be the size of the flash page (the same as `F::Page`)?

alistair23, 2022-11-09:
The idea would be to read from offset until the page boundary. Again it is never used though

The `2048` is the `const S`. From the documentation

```rust
/// The size of the regions (pages) must be the smallest size that can be
/// erased in a single operation. This is specified as the constant `S`
/// when implementing `FlashController` and `TicKV` and it must match
/// the length of the `read_buffer`.
```

alexandruradovici, 2022-11-10:
Considering that we use the `Flash` HIL underneath, should 2048 be the actual size of `F::Page`?

alistair23, 2022-11-10:
Yeah, it probably should

teonaseverin, 2022-11-10:
What about modifying `TickFSFlastCtrl` in such a way that it relates to the page size?

```rust
pub struct TickFSFlastCtrl<'a, F: Flash<Page = &'static mut [u8; S]> + 'static, const S: usize> ...

impl<'a, F: Flash<Page = &'static mut [u8; S]>, const S: usize> tickv::flash_controller::FlashController<S> for TickFSFlastCtrl<'a, F, S> ...
```

@alistair23 do you think this would be correct?

Is the name `TickFSFlastCtrl` a typo for `TickFSFlashCtrl` ?

alistair23, 2022-11-10:
Yeah, I think that's worth doing for Tock. Also that's a typo

twilfredo, 2022-11-11:
> What about modifying `TickFSFlastCtrl` in such a way that it relates to the page size?
>
> ```rust
> pub struct TickFSFlastCtrl<'a, F: Flash<Page = &'static mut [u8; S]> + 'static, const S: usize> ... [...]

Typo fixed in https://github.com/tock/tock/pull/3323

bradjc, 2023-05-30:
This is addressed in #3458

closed 2023-06-15 by bradjc