tock/tock #3340: Grant entry unsafe in the face of process restart

LawrenceEsswood, 2022-12-01:
Process restart effectively deallocates the objects stored in grant regions, but does not respect the fact that there may exist references to them.

Pseudo-example:

```
some_grant.enter(app_id, |grant_data, kern_data| {
  // Now cause a restart, either indirectly...
  trigger_app_fault();
  // Or directly if we are in the kernel (e.g. IPC module), or have capabilities::ProcessManagementCapability...
  kernel.process_map_or_external((), app_id, |proc| { proc.set_fault_state();} ,  my_cap);
  // Now grant_data dangles to re-allocable grant memory.
  // Call another capsule to get them to allocate their grant to overlap where ours used to be. We would probably loop over all 
  // procs to find our new restarted one, given our app_id is no longer valid.
  call_other_capsule();
  // Type confusion for fun and profit
  grant_data.field = anything I like
});
```

It is also slightly dodgy that in the context of the ``process_map_or_external`` the ``app_id`` will no longer match the ``proc``. We are lucky that we don't really re-allocate the proc object. If we did, for example if the restart were not in place, we would also have a safety issue there.

I notice ``grant_ptrs_reset`` is unsafe, but no other part of termination is. Unsafe should likely be further propagated up the call chain, given how unsafe de-allocation really is.

IMO I don't think the error here is failure to check that grants were open or not. I think the real issue is that restart can be called from anything but the kernel in its main loop (thus giving one place to reason that any ephemeral objects like &dyn Process and GrantData<T> etc). ``trigger_fault`` should just set state to something like ``restarting``, and then the kernel should make the actual call to restart from a less deeply nested context. It would also sidestep any panics that fixing this would otherwise have.

Also, the T in GrantData<T> is not actually dropped on deallocation. Somewhat unrelated, but I thought I would throw it in the ring given I was poking around this area. Some capsules may care that their T's are being ripped from under them, to update queues and the like.

brghena, 2022-12-01:
I think the idea here was that if a process was valid when the grant was entered, it will remain valid the entire time. The process is not running, so it definitely can't fault itself. Hopefully capsules can't fault processes either, although I'm less sure about that.

The solution of having app cleanup only occur as part of the kernel loop sounds right to me.

LawrenceEsswood, 2022-12-01:
I think that second point is the sticker. They can if they have ``ProcessManagementCapability`` or can invoke other functionality in the kernel can.

hudson-ayers, 2022-12-01:
Yeah, for upstream Tock boards today this basically means that someone could write unsound code in the process_console or IPC capsules without invoking `unsafe`. I think I agree with the suggested solution as well