tock/tock #3349: Support for Xtensa Arch

Herkulisch, 2022-12-08:
# Background
Hey I've read the docs and it said that ports for other architectures are welcome but one should ask via mail. 
I thought it would be better to ask this in an issue. 
So I would like to use Tock to build a small OS for the ESP32-PICO-D4, which gets used inside the [Watchy Project](https://github.com/sqfmi/Watchy). 
I had a Pebble Smartwatch before they were bought by FitBit and then shut down and I would like to build a similar experience in my Free Time. 
But instead of using FreeRTOS like the Pebble did, I would like to use your project and of course the Rust language.

Unfortunately only ESP32s with RISC-V architecture are supported. 
So I thought why not implement the necessary features for the Xtensa architecture while learning a lot in the progress!

# Actual Question
Is Tock platform agnostic enough that I could it working on the Xtensa architecture? 
Or do you guys think it is not worth the afford right now?
If that is not the case this could also be something that can be listed in the Wiki under the [Project Ideas Entry](https://github.com/tock/tock/wiki/Project-Ideas-for-Tock).

# Thanks
I'd like to thank you guys in advance and I am excited to hear from you guys.
If you do not respond to the issue in a week I'll probably bother you via Mail or Slack with a copy paste version of this. 😂

alexandruradovici, 2022-12-08:
Hi, thank you for the interest in porting Tock to a new architecture. Tock should be platform agnostic, you can start by taking a look at the `arch/` crates.

I thought about this some time ago and these are the things that I saw as blockers:

- at the time Xtensa was not officially supported by Rust, it was a fork. I had a difficult time building it. I think ESP is doing some steps in this direction now and they have Rust working more or less out of the box for their boards (bare metal)
- ESP uses a lot of binary blobs for their code (WiFi, BLE peripherals are not event documented in their manual). I tried including one of them, just as en experiment into Tock, and gave up due to lack of documentation

I suggest you take a look at https://github.com/esp-rs, the repository of ESP for Rust.

jrvanwhy, 2022-12-08:
I'm not familiar with Xtensa, but as far as I can tell (from some quick searching):

1. It's a reasonably-popular 32-bit embedded architecture
2. `rustc` has support for it

so it seems like a reasonable architecture for Tock to target.

You will need to extend [TRD 104 (Tock's syscall ABI)](https://github.com/tock/tock/blob/master/doc/reference/trd104-syscalls.md) to Xtensa. You'll also need to write all the architecture-specific code.

One challenge is that I'm not sure anyone on the Tock core WG (the developers who lead Tock development) has Xtensa experience. If one of us does, hopefully they'll speak up. Communication, debugging, and code ownership in general will be more difficult/complicated if none of us has an Xtensa dev board.

alexandruradovici, 2022-12-08:
> One challenge is that I'm not sure anyone on the Tock core WG (the developers who lead Tock development) has Xtensa experience. If one of us does, hopefully they'll speak up. Communication, [...]

If you start working on this, I think I might be able to connect you to developers that worked at the ESP bare metal Rust port.

brghena, 2022-12-08:
There's also the question of whether it's suitable for Tock. https://github.com/tock/tock/blob/master/doc/Porting.md#is-tock-a-good-fit-for-my-hardware

You should be fine on the 32-bit support. And it looks like your particular chip has 448 KB of ROM and 520 KB of SRAM, so memory is good.

But the memory protection mechanism will be important too. From a little googling it looks like the Xtensa architecture has support for an MMU or for some kind of "region protection". It looks like the ESP32 has several different options (actually too many, looks complicated), but I suspect it'll be sufficient. https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#mpummu

Herkulisch, 2022-12-08:
> There's also the question of whether it's suitable for Tock. https://github.com/tock/tock/blob/master/doc/Porting.md#is-tock-a-good-fit-for-my-hardware
> 
> You should be fine on the 32-bit support. And it looks like your particular chip has 448 KB of ROM and 520 KB of SRAM, so memory is good.
> 
> The memory protection mechanism will be important too. From a little googling it looks like the Xtensa architecture has support for an MMU or for some kind of "region protection". It looks like the ESP32 has several different options (actually too many, looks complicated), but I suspect it'll be sufficient. https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#mpummu

Yep I also had a look at the requirements beforehand and the MMU part was the only thing I was not sure about, so I should investigate that further. But, like you, I think it should be sufficient.

Also Thanks for all the quick replies guys. 😄

Herkulisch, 2022-12-08:
> I'm not familiar with Xtensa, but as far as I can tell (from some quick searching):
>
>     1. It's a reasonably-popular 32-bit embedded architecture
>
>     2. `rustc` has support for it
>
>
> so it seems [...]

Thank you for guiding me into the right direction! I will have a look at his 😄

bradjc, 2022-12-16:
Porting to an architecture is challenging because it is not always obvious how to map Tock's expectations of context switches to what the actual hardware provides. The core of the port is implementing the `UserspaceKernelBoundary`. You can see the risc-v port here: https://github.com/tock/tock/blob/master/arch/rv32i/src/syscall.rs and https://github.com/tock/tock/blob/master/arch/rv32i/src/lib.rs

Having worked on the initial port to riscv, I'll add two comments 1) since Tock supports both arm and riscv, and architecturally they are fairly different, I believe our abstractions should make the port entirely contained to the arch/ directory, and 2) I would recommend sketching out a plan to implement all of the traits for `UserspaceKernelBoundary` before actually writing code. In my experience other embedded code bases that use context switches don't quite use them the way that Tock does and so it can be difficult to find good example code to reference from.

Also, because there can be a fair bit of subtlety in the low level arch code we want to see it commented like the riscv port.

elasticdotventures, 2023-09-30:
Just a +1 for ESP32 support.