tock/tock #3354: Update the app panic printout to reflect "make lst" only applies to libtock-c apps

potto216, 2022-12-11:
Currently if an app panics, the panic printout includes the following message:
```
...
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug, run `make lst` in the app's folder
and open the arch.0x30060.0x20008000.lst file.
```
The line "To debug, run `make lst` in the app's folder" only applies to libtock-c apps and not libtock-rs apps. To reduce confusion I propose changing the message to read: "If this is a libtock-c app, then to debug, run `make lst` in the app's folder".

This message is found in [process_standard.rc] (https://github.com/tock/tock/blob/master/kernel/src/process_standard.rs#L1278)
https://github.com/tock/tock/blob/4bb246fe061e1fde9d595e76c1e1994b8ccc224d/kernel/src/process_standard.rs#L1278.

If you accept PRs I can make the change.

alistair23, 2022-12-11:
A PR is already open for this: https://github.com/tock/tock/pull/3351 :)

potto216, 2022-12-11:
Closing because it is a duplicate of https://github.com/tock/tock/pull/3351

closed 2022-12-11 by potto216