tock/tock #3356: nrf52 cdc issues with new board

dwagenk, 2022-12-17:
Hi,

I've been trying to port tock to the [Adafruit ItsyBitsy nRF52840 Express](https://learn.adafruit.com/adafruit-itsybitsy-nrf52840-express/), mostly copying the already supported Arduino Nano 33 BLE. My port currently lives at [codeberg:dwagenk/tock](https://codeberg.org/dwagenk/tock/src/branch/WIP/feature/nrf52-itsy-bitsy-support). The [bootloader](https://codeberg.org/dwagenk/tock-bootloader/src/branch/feature/nrf52-itsy-bitsy-support) is basically ready, but not yet submitted as a PR.

I'm running into issues with the console and panic handler via CDC. This confuses me, since everything related to CDC is configured completely identical to the nano33ble board. CDC in the bootloader works fine and I can flash the kernel and apps via it without any problems.

- With console and panic via CDC (`847e0f149 Initial support for itsybitsy_nrf52840_express board`) :
  - No output on console when panic occurs (e.g. by invoking the `panic` command in the process console)
  - Panic LED does not blink
  - `list` command on the process console also locks up the board in the same way
- With console via CDC but panic via UART pins (`e2dbe5204 WIP: itsybitsy: use real uart for panic, not cdc`):
  - running `panic` command in process console works as expected, panic output on UART and panic LED blinks
  - `list` command on the process console still locks up the board, no panic output on UART and no blinking panic LED
- With console and panic via UART pins (`d5b84a76 WIP: itsybitsy: completely switch to uart instead of cdc`):
  - works as expected

Any ideas for troubleshooting? The Difference to the nano33ble board is minimal
- PID/VID for USB CDC uses the values as the original Adafruit ItsyBitsy
- removed proximity, temperature and humidity sensor support
- only 1 LED instead of 3
- less ADC channels
- added support for the button

So there really is not much that I could have messed up. I still guess It's related to my port, but since I don't have the nano33ble board available I'd be happy if a user of that board can check that it works as expected there.

Any help is welcome!

hudson-ayers, 2022-12-18:
I ran some quick tests tonight on my nano33, expecting to swiftly verify that panic works as expected, and found that something seems to have completely broken since the 2.1 release -- I get no console output over CDC at all. I don't have time to debug further tonight but wanted to give you a signal that it is indeed possible the bug you are seeing is not specific to your port

dwagenk, 2022-12-18:
Thanks for testing!

I realized I can compile and run the nano33ble binary on my target. The CDC related stuff is identical and the missing sensors don't matter for testing.

With the in tree nano33ble support and the toolchain from `shell.nix` I started bisecting. I used the `list` command in the process console as a test (good: lists processes, bad: crashes board).

For this problem the first `bad` commit is:
bc17e7e6c `update nightly oct 2022`

But I encountered the problem you described as well (no CDC at all or more accurately board crashing early, since the blink application I have loaded does not seem to run). I'll try bisecting that one now.

dwagenk, 2022-12-18:
>> I get no console output over CDC at all.

> But I encountered the problem you described as well (no CDC at all or more accurately board crashing early, since the blink application I have loaded does not seem to run). I'll try bisecting that one now.

bisect showed this to be the first `bad` commit
bab2b62ae `boards: nano33ble: use i2c mux component`

dwagenk, 2022-12-18:
Just one additional note: with the nano33ble kernel flashed on my board I didn't find any commit, that had working panic via CDC. Console via CDC worked in most of these combinations, but panic did not output anything via CDC and did not even panic-blink the LED (that part was only somewhat tested by me, since it required modification of the `led_kernel_pin` for every tested commit which I probably missed on some of them)

Tested commits include:
- 163875411 `Synchronous panic output over USB for nano33ble`
- 6ade39d27 `(tag: release-2.0) Merge #2799`
- ffa5ce02b `(tag: release-2.1) Merge #3194 #3195`
- 990ff6068 `(upstream/master) Merge #3351`console didn't work at all, see bisection results in previous comments

hudson-ayers, 2022-12-27:
https://github.com/tock/tock/pull/3362 fixes the first bug for the nano33. It looks like panic output over CDC is also broken for the nano33 on current master, however. Working on debugging that. It definitely worked at 1638754. Something else seems to have broken with the process console since then as well (using the process console works fine with no apps loaded, but with any apps loaded it panics (and then I get no panic output because that is also broken).

bradjc, 2023-08-18:
I gave fixing the cdc panic print a try. I have no idea what is wrong.

I tried using cdc for panics (and only panics) on the nrf52840dk with the code from the nano33, and it works ok. It prints some of the panic print but eventually stops. The amount that prints is different on every panic.

The closest I got to something is it seems like other interrupts are interfering somehow. But I added

```rust
cortexm4::nvic::disable_all();
cortexm4::nvic::clear_all_pending();
let n = cortexm4::nvic::Nvic::new(39);
n.enable();
```

to the start of the panic function and that didn't fix it.

I tried adding `wfi()` which didn't help.

As far as I can tell at some point we call `cdc.transmit_buffer()` and we just never get the `transmitted_buffer()` callback. I can't trace anything in the code to understand why that might be.