tock/tock #3375: All drivers should return `Success` rather than `SuccessU32` for command 0

hudson-ayers, 2023-01-11:
https://github.com/tock/tock/blob/master/doc/reference/trd104-syscalls.md#431-command-identifier-0 specifies that command 0 should always return `Success`. However, at least the following 4 drivers do not follow this direction:

- ADC https://github.com/tock/tock/blob/500caa5e882b7efe40158849fb489205f07c6a2e/capsules/src/adc.rs#L1149
- GPIO https://github.com/tock/tock/blob/500caa5e882b7efe40158849fb489205f07c6a2e/capsules/src/gpio.rs#L198
- LED https://github.com/tock/tock/blob/500caa5e882b7efe40158849fb489205f07c6a2e/capsules/src/led.rs#L95
- PWM https://github.com/tock/tock/pull/3367

I recall there being a desire for userspace to be able to build helper functions around command 0 with the assumption of a fixed Success variant for that command, so we should update these 4 drivers by adding a new command number to check the number of resources for that particular driver. This will require matching updates to `libtock-rs` and `libtock-c` for each

jrvanwhy, 2023-02-13:
ADC and LED are stable capsules, and changing this call is a breaking change, so wouldn't this require a version bump to 3.0?

Or is this okay because "TRD 104 says it should be this way"?

ppannuto, 2023-02-16:
I would like to think the "stable" does indeed mean "stable" for Tock 2.x, [we currently say](https://github.com/tock/tock/tree/master/doc/syscalls):
> Each driver type that has been allocated a permanent driver number is listed in the tables below. The "2.0" column indicates whether the driver has been stabilized or not (a "✓" indicates stability) in the Tock 2.0 release.

I think a path forward might be to mark driver `0x00002` as deprecated, and pending removal in the Tock 3.x release, and have the revised driver from #3397 be a new `0x0000a` driver, available from Tock 2.2+, and as the new stable LED interface for Tock 3.x?

Downstream platforms concerned about code size could, of course, omit the `0x00002` driver in their internal implementations to avoid duplicate copies of the LED driver.

---

LED happens to be stateless, so in this case we could likely have correct operation simply keeping two copies of the LED capsules around (caveat: two Rust references to the same MMIO region). The better thing would probably be to write a "legacy LED" capsule or some such that lives underneath `0x00002` and forwards / translates commands to the "real LED" capsule, wired directly to `0x0000a`?

hudson-ayers, 2023-02-16:
I don't think we want to have duplicate drivers for LED/ADC, that feels like a potential source for really weird bugs and generally a waste of resources. I would rather just go ahead and bump to 3.0 than do that -- there is no reason each major version bump has to be because of a major change.

That said, @jrvanwhy, how important is it that we make this change? I might advocate for just removing the language from TRD104 over going through the effort of a major version bump if it wouldn't be that valuable, and I recall libtock-rs being the motivating reason for it.

jrvanwhy, 2023-02-16:
> That said, @jrvanwhy, how important is it that we make this change? I might advocate for just removing the language from TRD104 over going through the effort of a major version bump if it wouldn't be that valuable, and I recall libtock-rs being the motivating reason for it.

IIRC, the purpose of that language in TRD104 is to allow a function like the following to exist:

```rust
// Works for any driver number
fn check_if_driver_present(driver_num: u32) -> bool {
    command(driver_num, 0, 0, 0).is_success()
}
```

where `is_success` only returns true for a success without any payload.

I don't think that `check_if_driver_present` is very valuable (I can't think of a use case), so I don't think that bumping to 3.0 is worth it for this change. Seems more like something we should fix in 3.0 *when we have a reason for 3.0 to happen*.

jrvanwhy, 2023-02-16:
> That said, @jrvanwhy, how important is it that we make this change? I might advocate for just removing the language from TRD104 over going through the effort of a major version bump if it wouldn't be that valuable, and I recall libtock-rs being the motivating reason for it.

Oh and to be clear: there is nothing specific to `libtock-rs` about this. IIRC `check_if_driver_present` does not exist in `libtock-rs`, but it may exist in `libtock-c`.

hudson-ayers, 2023-02-17:
> ADC and LED are stable capsules, and changing this call is a breaking change, so wouldn't this require a version bump to 3.0?

Wait, how do we mark capsules as stable again? The TRDs for those capsules are still draft, not finalized

bradjc, 2023-02-17:
> > ADC and LED are stable capsules, and changing this call is a breaking change, so wouldn't this require a version bump to 3.0?
> 
> Wait, how do we mark capsules as stable again? The TRDs for those capsules are still draft, not finalized

A checkmark in this document: https://github.com/tock/tock/blob/master/doc/syscalls/README.md

hudson-ayers, 2023-02-22:
After some discussion in last weeks core call, we decided that all stabilized drivers should stay as they are until whenever we decide to do a Tock 3.0. However, other drivers can change. I submitted a PR adding a warning to the three stabilized drivers (LED / Button / ADC): https://github.com/tock/tock/pull/3402 . 

The remaining drivers that need to be changed are:
- [ ] gpio
- [ ] pwm
- [ ] analog_comparator
- [ ] seven_segment
- [ ] read_only_state

I am editing the issue description to reflect this

alevy, 2023-08-17:
Coming back to this thanks to #3613....

How bad of an option is it to change the TRD to allow for both `success` and `success_with_u32`, where the optional additional return value means there are this many instances of the thing (as in GPIO and LED).

I believe this would require no changes to stable drivers and minimal change to the hypothetical `is_success()` function (it would be `>=128` instead of `==128`).

In fact we could loosen TRD104 to just say that any success response means the driver exists, that the returned values, if any, can be driver-specific, and that the only other guarantee is that command number `0` doesn't do I/O (it's cheap and doesn't have side effects).

bradjc, 2023-08-17:
> How bad of an option is it to change the TRD to allow for both `success` and `success_with_u32`, where the optional additional return value means there are this many instances of the thing (as in GPIO and LED).

But what is the advantage? That wouldn't change anything in the short term (code is already there), and it doesn't save us that much effort for 3.0 (again, code is already there in a PR). Seems like we need to decide what we want.

I question the utility of APIs for "how many leds/adc channels/gpio pins/etc are on this board". They are _great_ for demo apps (which is almost certainly why we added them in the first place). But thinking long term: either you are trying to write a cross-platform Tock app where you can't rely on each board having the same number of some resource and it would be hard to know what to even do with N leds or M adc channels. Or you are writing for a very specific board, and you don't need to query how many of a resource because you already know.

What we need is for user space to have a semantic understanding. What are these LEDs, and how are they shown to users? What is this ADC channel connected to?

I think being able to check for a driver is very useful and we should keep the TRD104 approach as currently written. We should also allow capsules to expose counts. But we should keep those separate.