tock/tock #3378: Nothing happens after flashing blink to board STM32F429 Nucleo-144

Eplankton, 2023-01-16:
I use the board which supported by Tock according to the hardware manual
my environment:
```
Ubuntu-22.04
OpenOCD 0.11.0
ST-LINK V2-1 (on board)
```
It appears to behave normally during flashing kernel:

[image]

and also succeed in flashing `blink` to the board:

[image]

However, after that, nothing happen, my LEDs didn't blink at all.

alexandruradovici, 2023-01-17:
Due to the flash layout (uneven sectors), you might not be able to flash apps using tockloader to this board.

A few debug ideas are:
- flash only the kernel using `make install`
- run `tockloader listen` and see if you can use the process console (it writes something like `tock$` and you should be able to write some commands)
- flash an app using tockloader and then try listing the apps with tockloader (`tockloader info`)
- run `tockloader listen` and execute the `list` command
- if there are no processes running, enable the kernel process loading debug messages by adding `features = ["debug_load_processes"]` to *boards/nucleo_f429zi/Cargo.toml* to the *kernel* dependency

hudson-ayers, 2023-01-17:
I would start by trying to build blink locally and flash it, rather than using the online "blink" TAB, which may be out of date

Eplankton, 2023-01-18:
> Due to the flash layout (uneven sectors), you might not be able to flash apps using tockloader to this board.
>
> A few debug ideas are:
>
>     * flash only the kernel using `make install`
>
>     * run [...]

Got something like this:
```
zhao@ nucleo_f429zi $ ./listen.sh 
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] No serial port with device name "tock" found.
[INFO   ] Found 1 serial port.
[INFO   ] Using "/dev/ttyACM0 - STM32 STLink - ST-Link VCP Ctrl".
[INFO   ] Listening for serial output.
�Initialization complete. Entering main loop
Loading processes from flash=0x08040000-0x0807FFFF into sram=0x200042C8-0x2002FFFF
Loading process from flash=0x08040000-0x0807FFFF into sram=0x200042C8-0x2002FFFF
No more processes to load: Could not find TBF header.
tock$ 
```

Eplankton, 2023-01-18:
> Due to the flash layout (uneven sectors), you might not be able to flash apps using tockloader to this board.
>
> A few debug ideas are:
>
>     * flash only the kernel using `make install`
>
>     * run [...]

By the way, what commands can I input to `tock$` ? , only `list` ?

Eplankton, 2023-01-19:
> 由于闪存布局（扇区不均匀），您可能无法使用 tockloader 将应用程序闪存到此板。
> 
> 一些调试思路是：
> 
>     * 仅闪存内核使用 `make install`
> 
>     * 跑步 `tockloader listen`看看你是否可以使用流程控制台（它写的是 `tock$`你应该能够写一些命令）
> 
>     * 使用 tockloader 刷新应用程序，然后尝试使用 tockloader 列出应用程序（ `tockloader info`)
> 
>     * 跑步 `tockloader listen`并执行 `list`命令
> 
>     * 如果没有进程在运行，则通过添加启用内核进程加载调试消息 `features = ["debug_load_processes"]`到 _boards/nucleo_f429zi/Cargo.toml _到 _内核 _依赖

I wonder whether there's an effective way to flash 'blink' to my board......

hudson-ayers, 2023-01-19:
Sorry, I haven't used this board before, I think my initial comment was not helpful.

This board does not correctly support tockloader (as Alex mentioned), but its README describes an alternate approach for flashing apps: https://github.com/tock/tock/tree/master/boards/nucleo_f429zi#flashing-app

I suggest you give that a try

Eplankton, 2023-01-20:
> 

Thanks for your suggestion,I'll have a try later. And I believe the problem is that ST has a special config on their ST-LINK on board, so OpenOCD can't correctly flash, may be adapt to certain version of openocd with fix it.

Eplankton, 2023-01-20:
> 
I succeed in flashing `blink` to my board, but only the blue-led blinks:
```
zhao@ nucleo_f429zi $ ./info.sh 
[INFO   ] Using settings from KNOWN_BOARDS["nucleof4"]
tockloader version: 1.9.0
[STATUS ] Showing all properties of the board...
Apps:
┌──────────────────────────────────────────────────┐
│ App 0                                            |
└──────────────────────────────────────────────────┘
  Name:                  blink
  Enabled:               True
  Sticky:                False
  Total Size in Flash:   2048 bytes
  Address in Flash:      0x8040000
    version               : 2
    header_size           :         52         0x34
    total_size            :       2048        0x800
    checksum              :              0x6e5075d7
    flags                 :          1          0x1
      enabled             : Yes
      sticky              : No
    TLV: Main (1)
      init_fn_offset      :         41         0x29
      protected_size      :          0          0x0
      minimum_ram_size    :       4604       0x11fc
    TLV: Package Name (3)
      package_name        : blink
    TLV: Kernel Version (8)
      kernel_major        : 2
      kernel_minor        : 0
      kernel version      : ^2.0

[INFO   ] Finished in 14.527 seconds
```

And now some painc happens:
```
zhao@ nucleo_f429zi $ ./listen.sh 
[INFO   ] No device name specified. Using default name "tock".
[INFO   ] No serial port with device name "tock" found.
[INFO   ] Found 1 serial port.
[INFO   ] Using "/dev/ttyACM0 - STM32 STLink - ST-Link VCP Ctrl".
[INFO   ] Listening for serial output.
�
panicked at 'Kernel HardFault.
        Kernel version 2.1+
        r0  0x200040ac
        r1  0x0
        r2  0x801dde0
        r3  0x3
        r12 0x3c
        lr  0x800b91b
        pc  0x800b3e0
        prs 0x61000000 [ N 0 Z 1 C 1 V 0 Q 0 GE 0000 ; ICI.IT 0 T true ; Exc 0-Thread Mode ]
        sp  0x20001dc8
        top of stack     0x20002000
        bottom of stack  0x20000000
        SHCSR 0x0
        CFSR  0x80000
        HSFR  0x40000000
        Instruction Access Violation:       false
        Data Access Violation:              false
        Memory Management Unstacking Fault: false
        Memory Management Stacking Fault:   false
        Memory Management Lazy FP Fault:    false
        Instruction Bus Error:              false
        Precise Data Bus Error:             false
        Imprecise Data Bus Error:           false
        Bus Unstacking Fault:               false
        Bus Stacking Fault:                 false
        Bus Lazy FP Fault:                  false
        Undefined Instruction Usage Fault:  false
        Invalid State Usage Fault:          false
        Invalid PC Load Usage Fault:        false
        No Coprocessor Usage Fault:         true
        Unaligned Access Usage Fault:       false
        Divide By Zero:                     false
        Bus Fault on Vector Table Read:     false
        Forced Hard Fault:                  true
        Faulting Memory Address: (valid: false) 0xE000ED34
        Bus Fault Address:       (valid: false) 0xE000ED38
', arch/cortex-m/src/lib.rs:495:5
```

hudson-ayers, 2023-01-20:
That is a very unusual panic...it suggests that the processor is attempting to access a coprocessor (which this chip does not have). I don't immediately have any idea as to why that might be happening.

Eplankton, 2023-01-21:
> 

I can imagine that the problem may occurs at the compiler or openocd.cfg which configures the system for another chip...

hudson-ayers, 2023-01-23:
Which chip/board are you flashing, exactly?

Eplankton, 2023-02-13:
> Which chip/board are you flashing, exactly?

Well, I wondering whether this problem can be fixed or not now....still waiting, while I believe there's something wrong with my the board itself.

hudson-ayers, 2023-02-14:
Unfortunately this going to be tough for me to debug since I do not have that board -- hopefully a Tock contributor who does can chip in

Eplankton, 2023-02-14:
> Unfortunately this going to be tough for me to debug since I do not have that board -- hopefully a Tock contributor who does can chip in

OK, and thanks for your concern, sir.

bradjc, 2023-08-18:
Please re-open with any new updated.

closed 2023-08-18 by bradjc