tock/tock #3383: License headers: interpretation of "blank line"?

jrvanwhy, 2023-01-24:
[`trd-legal.md`](https://github.com/tock/tock/blob/master/doc/reference/trd-legal.md) says:

> License and copyright information SHOULD have at least one (1) blank line separating it from any other content in the file.

When I implemented the license checker, I interpreted this as prohibiting `//` and `#` symbols in the blank line. However, four files in the repository have a `//` line:

[`chips/stm32f4xx/src/can.rs`](https://github.com/tock/tock/blob/master/chips/stm32f4xx/src/can.rs)
[`boards/components/src/can.rs`](https://github.com/tock/tock/blob/master/boards/components/src/can.rs)
[`kernel/src/hil/can.rs`](https://github.com/tock/tock/blob/master/kernel/src/hil/can.rs)
[`capsules/src/can.rs`](https://github.com/tock/tock/blob/master/capsules/src/can.rs)

We need to clarify which interpretation of "blank line" is correct. Should we remove the `//` symbols from the above four files, or adjust the license checker to accept the `//` symbols?

hudson-ayers, 2023-01-24:
I would vote for it being a truly blank line (e.g. no `//` etc.) but don't feel strongly about this.

brghena, 2023-01-27:
No one felt super strongly about this, and most of us didn't care at all. The conversation led to a decision to be permissive whenever possible, so accepting both blank lines and commented lines seems like the right choice.

commit 8538bd63 (2023-02-07): Merge #3386 #3387

3386: Add core WG notes for `2023-01-27`. r=jrvanwhy a=jrvanwhy

3387: `licensecheck`: Allow an empty comment line after the header. r=jrvanwhy a=jrvanwhy

### Pull Request Overview

`trd-legal.md` says:

> License and copyright information SHOULD have at least one (1) blank line separating it from any other content in the file.

When I implemented the license checker, I interpreted that as disallowing an empty comment. However, a few files in this repository have an empty comment line after the header instead. This changes the license checker to accept either a truly blank  [... cut]
PR #3387 (2023-02-07): `licensecheck`: Allow an empty comment line after the header.
### Pull Request Overview

`trd-legal.md` says:

> License and copyright information SHOULD have at least one (1) blank line separating it from any other content in the file.

When I implemented the license checker, I interpreted that as disallowing an empty comment. However, a few files in this repository have an empty comment line after the header instead. This changes the license checker to accept either a truly blank line or an empty comment line after the license header.

Closes #3383

### Testing Strategy

I deleted the `*` li [... cut]
closed 2023-02-07 by bors[bot]