tock/tock #3412: Blink example not working on Raspberry Pi Pico (W)

J0hn-Lawrence, 2023-03-10:
<short summary of the problem>
Compiled and flashed Blink example C code onto Raspberry Pi Pico W via USB. No LED blinking.

I expected to see this happen:
After flashing via '$ APP="<path-to-tbf-file-of-blink-example>" make program' the Pi Pico restarts and the LED starts to blink.

Instead, this happened: 
After flashing via '$ APP="<path-to-tbf-file-of-blink-example>" make program' the Pi Pico restarts (as it should after flashing), but the LED on the board does not blink.

Hardware platform:
Raspberry Pi Pico W

Is this behavior expected? Did I do something wrong regarding the flashing progress? Any help is much appreciated!

Apologies if this is the wrong place for this issue. But I don´t know where else to ask about this.

alexandruradovici, 2023-03-10:
Hi, thank you for reporting this. Can you post here the output of `make program`?

J0hn-Lawrence, 2023-03-15:
Hi,
yes. Here is the output:

APP="/home/matix/Documents/repos/libtock-c/examples/c_hello/build/cortex-m0/cortex-m0.tbf" make program
    Finished release [optimized + debuginfo] target(s) in 0.02s
   text	   data	    bss	    dec	    hex	filename
  98308	      0	  13216	 111524	  1b3a4	/home/matix/Documents/repos/tock/target/thumbv6m-none-eabi/release/raspberry_pi_pico
arm-none-eabi-objcopy --update-section .apps=/home/matix/Documents/repos/libtock-c/examples/c_hello/build/cortex-m0/cortex-m0.tbf /home/matix/Documents/repos/tock/target/thumbv6m-none-eabi/release/raspberry_pi_pico.elf /home/matix/Documents/repos/tock//target/thumbv6m-none-eabi/release/raspberry_pi_pico-app.elf
elf2uf2-rs /home/matix/Documents/repos/tock//target/thumbv6m-none-eabi/release/raspberry_pi_pico-app.elf /home/matix/Documents/repos/tock/target/thumbv6m-none-eabi/release/raspberry_pi_pico-app.uf2

ptka, 2023-03-15:
My information is, that the onboard LED on the Pico W is connected to WLAN module and not GPIO25 as on the wireless Pico. For me the behavior looks correct as there's no issue in the blink example.

J0hn-Lawrence, 2023-03-16:
> My information is, that the onboard LED on the Pico W is connected to WLAN module and not GPIO25 as on the wireless Pico. For me the behavior looks correct as there's no issue in the blink example.

Thank you @ptka! Your information seams correct. 
  - The blink example does in fact work on Raspberry Pi Pico. 
  - It does not work on the Raspberry Pi Pico W. 
 I think this issue can be closed now, as Pico works and Pico W does not work as it is not supported. Thank you guys for clarifying.

closed 2023-03-16 by J0hn-Lawrence