tock/tock #3417: License checker should support block comments

jrvanwhy, 2023-03-31:
Linker scripts do not support line comments (`//`-style comments). The license checker should support linker scripts, and should therefore be adjusted to support block comments as well.

Before implementing, I'll check whether there are other file types in the repository that only support block comments and make sure the implementation works for those too.

jrvanwhy, 2023-04-01:
Okay, I've expanded this issue somewhat, as I've found other files the license checker doesn't work on.

Some of the `.jlink` files use Unix line endings, other use DOS (CRLF) line endings. Currently this breaks the license checker, as it was only designed for Unix line endings. I'm a bit torn on how to resolve this. My personal preference is to convert those files to Unix line endings, but I could also adjust the license checker to support CRLF files.

jrvanwhy, 2023-04-01:
Oh, here's another file type that only supports block comments: XML

hudson-ayers, 2023-04-06:
+1 for all Unix line endings

commit a27eabc6 (2023-05-17): Use a fallback syntect syntax rather than a custom Rust parser in the license checker.

The license checker uses the `syntect` crate to parse files wherever possible. `syntect` provides a number of syntax definitions for common language types. However, we have some files that `syntect` does not recognize. Currently, the license checker falls back to a hand-coded parser if it encounters a file type that `syntect` does not recognize. This PR removes that hand-coded parser, and replaces it with a fallback `syntect` syntax. This removes the duplicate parsing logic, replacing it with an easier-to-e [... cut]
commit 6051a5eb (2023-05-17): Merge #3434

3434: Use a fallback syntect syntax rather than a custom Rust parser in the license checker. r=bradjc a=jrvanwhy

The license checker uses the `syntect` crate to parse files wherever possible. `syntect` provides a number of syntax definitions for common language types. However, we have some files that `syntect` does not recognize. Currently, the license checker falls back to a hand-coded parser if it encounters a file type that `syntect` does not recognize. This PR removes that hand-coded parser, and replaces it with a fallback `syntect` syntax. This removes the duplicate parsing  [... cut]
PR #3434 (2023-05-17): Use a fallback syntect syntax rather than a custom Rust parser in the license checker.
The license checker uses the `syntect` crate to parse files wherever possible. `syntect` provides a number of syntax definitions for common language types. However, we have some files that `syntect` does not recognize. Currently, the license checker falls back to a hand-coded parser if it encounters a file type that `syntect` does not recognize. This PR removes that hand-coded parser, and replaces it with a fallback `syntect` syntax. This removes the duplicate parsing logic, replacing it with an easier-to-ex [... cut]
commit 33b250a9 (2023-06-16): Add license headers to `*.xml` files.

AFAIK, these are the last files in the repository that should have license headers but don't, so this PR should resolve #3417.
closed 2023-06-16 by bradjc