tock/tock #3432: Support for the RISC-V LoFive Rev 1 board

bobbbay, 2023-05-07:
The [LoFive Rev. 1 ](https://media.digikey.com/pdf/Data%20Sheets/GroupGets%20PDFs/LoFive_R1_Web.pdf)is an e310x board developed as a smaller alternative to the HiFive Rev. B. Since this library supports to Rev. B., are they interchangeable/can I load Tock onto a Lofive without any problems?

alistair23, 2023-05-07:
You can give it a try :)

You will probably need to modify the board file to change the GPIO mappings and exposed ports.

In general though the e310x isn't a good fit for Tock. The board is just too small to fit Tock and applications

bradjc, 2023-05-17:
I would expect basic apps (ie blink/print) to just work. Updating pin mappings is probably not too hard.

But yeah, with so little RAM on that MCU it's hard to do anything interesting with Tock.

bobbbay, 2023-05-17:
Okay, makes sense. I'll close this for now then, unless any new developments come up. Thanks.

closed 2023-05-17 by bobbbay