tock/tock #3442: RFC: ProcessFaultPolicy: per-process policy or global policy?

bradjc, 2023-05-22:
For historical and technical reasons, the `ProcessFaultPolicy` is stored with the per-process state.

```rust
pub trait ProcessFaultPolicy {
    /// Decide which action the kernel should take in response to `process`
    /// faulting.
    fn action(&self, process: &dyn Process) -> process::FaultAction;
}
```

```rust
pub struct ProcessStandard<'a, C: 'static + Chip> {
    ...

    /// How to respond if this process faults.
    fault_policy: &'a dyn ProcessFaultPolicy,

    ...
}
```

On one hand, this makes sense: the kernel may want different policies for different processes.

However, I see two issues with this approach:

1. With process credential checking, the `ProcessStandard` object is created _first_ (with a reference to the fault policy stored), and _then_ the credentials are checked. This means there isn't a way to set the fault policy based on credentials.
2. We now have the `KernelResources` struct which is the defacto configuration struct for Tock kernels. Since `ProcessFaultPolicy` isn't in it, we have then all of the kernel policies/config in one place, except for the process restart policy. This is confusing.

### Alternative

The alternative is pretty straightforward: move `ProcessFaultPolicy` into the `KernelResources` struct.

Now, that would mean there is only a single policy for all processes, and that the policy itself would have to identify different processes and act accordingly.

#### Advantages

1. All configuration/policies for a kernel are in the same place.
2. The logic for determining which policy to follow only executes after a fault, not unconditionally when the board starts.

#### Challenges

1. We use the fault policy in one logical place: during context switches. Since `KernelResources` is only available within the main kernel loop, this is ok, we just have to pass in the policy to the process code which decides if the process has faulted or not. However, we also have more debugging-esque uses, specifically in the process console where we can on-demand cause an app to fault. This function is no longer callable, because it doesn't have access to the `KernelResources` object and therefore cannot provide a reference to the fault policy.

    To manage this, I think we would have to create a new process state `ProcessFaulting` which, on the next kernel loop, would cause the process to actually fault (but from the kernel loop so we have access to the kernel resources and fault policy.

### Proposal

We move to the alternative design to keep all configuration and policies in one place. Since fault policies should be able to use credentials, I think real-world fault policies are already going to be generic (i.e. one exists for every app, and that policy checks at fault-time if the app is trusted or not).

bradjc, 2023-05-22:
Basic implementation here: https://github.com/tock/tock/compare/process-fault-policy-global?expand=1

phil-levis, 2023-05-26:
Having per-process policies seems pretty important to me. Centralizing it and requiring the kernel to know about the individual processes seems like pushing further to bespoke kernels.

I don't see how 1) is a serious issue. Do we have use cases when the fault policy should be based on credentials? If there's an issue with when the process is set, could we add a `set` method to change it, based on other policies?

For 2, this makes sense to me if the policy is per-process. I'm not sure this is a kernel configuration issue, rather than a process one. For example, I could imagine a system call that allows a process to change its fault policy (e.g., from restart to stop, of course stuff like panic would require some special permissions).

bradjc, 2023-05-31:
Another proposal would be to separate three aspects of process fault handling into separate features:
1. When a process fault occurs, the kernel has to take some action in response (ie panic, restart, stop).
2. For each process, the kernel has to decide what the policy is going to be if the process faults.
3. Any sort of debugging or other side effects when a process faults.

To support this, we could have:

```rust
struct ProcessStandard {
    ...
    fault_action: Cell<FaultAction>,
    ...
}
```
which is what the kernel will do if that process faults.

Then we update the current trait:

```rust
trait ProcessFaultPolicy {
    /// Determine the process's fault action. The process may have requested
    /// a specific fault action, which this policy can choose to honor or not.
    fn determine_fault_action(process: Process, request: Option<FaultAction>) -> FaultAction;
}
```
to choose the fault action for a specific process.

And then to handle any debugging or side effects, we already have the `ProcessFault` callback (which was designed for fixing process faults, but could also handle something like calling `debug!()`).

One question with this approach, however, is when is `determine_fault_action()` called? To implement a policy like "restart 5 times then stop", we would need to change the `FaultAction`, so the `determine_fault_action()` function would have to run again.

bradjc, 2023-06-14:
Another reason to store the restart policy in `KernelResources` (or otherwise centrally) is that if we eventually support runtime loading of apps and we keep the restart policy in the process struct, we would still need to keep a reference to the restart policy around somewhere to use when creating the process struct.

bradjc, 2023-06-28:
Interestingly, the issue and proposed solution in #3340 would I think use the same infrastructure related to marking the process state as faulting but only actually doing the fault operation in the kernel loop as changing this mechanism would require.

bradjc, 2024-04-05:
I think we will leave this as a per-process setting.