tock/tock #3457: TicKV: Support for kernelspace and userspace

bradjc, 2023-05-26:
I'm experimenting with using TicKV in the kernel and as a way to share data between a host computer and a tock board (i.e. the host computer would use jtag to read and write the tickv kv store).

Currently, it looks like the capsule support for TicKV in Tock is aimed towards supporting userspace. What would it look like to support both userspace and in-kernel users of TicKV?

Some potential issues I see:
1. The kv_store capsule is both a wrapper on top of TicKV for tock userspace (defines a header format and uses StoragePermissions), as well as virtualizer. It seems like we would need to separate the virtualizer functionality.
2. Key collisions between userspace and the kernel.

alistair23, 2023-05-29:
You should be able to use `capsules/extra/src/kv_store.rs` in the kernel. Only the `KVSystemDriver` capsule (in `kv_driver.rs`) should be userspace specific.

You just need to have a kernel specific `StoragePermissions` for `kv_store.rs`. That could be as simple as reserving `0xFFFF_FFFF` for the kernel and not allowing applications to use/access those.

That doesn't fix key collisions, which are always a possible problem. One answer is to have separate instances (with different flash backing) for applications and kernel

bradjc, 2023-05-30:
How should this look? I ran into this because the constructor for `StoragePermissions` is `pub(crate)` so I couldn't create one in my main.rs file. Should the kernel just define a kernel storage permissions? Or should there be a capability-enforced constructor?

alistair23, 2023-06-06:
I think the kernel should define it's own `StoragePermissions`. The kernel is then in a position to enforce that no one else can use those permissions. Either in `get_storage_permissions()` or in the application loading.

I don't think it should be created in `main.rs`. The kernel permissions should be the same across all boards, not up to each board. Does that sound reasonable?

bradjc, 2023-06-06:
In that case I'm imagining something like:

in kernel crate:

```rust
fn get_kernel_storage_permissions(cap: &Capability) -> StoragePermissions {
   // return standard kernel permissions
}
```

alistair23, 2023-06-07:
Yeah, I think that seems like the way to go

hudson-ayers, 2023-07-11:
Closed by #3470

closed 2023-07-11 by hudson-ayers