tock/tock #3477: Wishlist: Explore error-on-format-failed

ppannuto, 2023-06-09:
Just [learned about this option](https://github.com/rust-lang/rustfmt/issues/4119#issuecomment-1401001368), `error_on_unformatted = true` for rustfmt. Might save folks from unexpected stuff like in #3475. Without having tested it at all, I am assuming there's a bunch of other minor things that would trip on this, so will take a procrastination-oriented afternoon someday :)

hudson-ayers, 2023-06-25:
Oh this seems great. I took a look at fixing this and currently there are 87 pieces of code in the repo that cannot be formatted correctly because of comment location. I think we should move all of those comments to places that do not break rustfmt and then add a `.rustfmt.toml` to the repo that only enables this option.

bradjc, 2023-06-26:
#3512

hudson-ayers, 2023-06-27:
Fixed by #3512

ppannuto, 2023-06-29:
Closed via #3512

closed 2023-06-29 by ppannuto