tock/tock #3478: kernel stack overflow panic after list command in microbit v2 board

itp18506, 2023-06-10:
I get this panic every time I execute list command with 1 and 2 apps running:
`lTemperature:                 31 deg C
Acceleration: X: 156 Y: -332 Z: -949
Magnetometer: X: 1227 Y: -1077 Z: 1586
Sound Pressure:              69

ist

panicked at 'kernel stack overflow', arch/cortex-m/src/lib.rs:636:13
	Kernel version release-2.1-1109-gf0501a03a

---| No debug queue found. You can set it with the DebugQueue component.

---| Cortex-M Fault Status |---
No Cortex-M faults detected.

---| App Status |---
𝐀𝐩𝐩: sensors   -   [Yielded]
 Events Queued: 0   Syscall Count: 126975   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Yield { which: 1, address: 0x0 }
 Completion Code: None
 Credential: None

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20007000═╪══════════════════════════════════════════╝
             │ Grant Ptrs      120
             │ Upcalls         320
             │ Process         920
  0x20006AB0 ┼───────────────────────────────────────────
             │ ▼ Grant         152
  0x20006A18 ┼───────────────────────────────────────────
             │ Unused
  0x200061DC ┼───────────────────────────────────────────
             │ ▲ Heap         1608 |   3716               S
  0x20005B94 ┼─────────────────────────────────────────── R
             │ Data            916 |    916               A
  0x20005800 ┼─────────────────────────────────────────── M
             │ ▼ Stack         544 |   2048          
  0x200055E0 ┼───────────────────────────────────────────
             │ Unused
  0x20005000 ┴───────────────────────────────────────────
             .....
  0x00044000 ┬─────────────────────────────────────────── F
             │ App Flash     16332                        L
  0x00040034 ┼─────────────────────────────────────────── A
             │ Protected        52                        S
  0x00040000 ┴─────────────────────────────────────────── H

  R0 : 0x00000001    R6 : 0x0004272C
  R1 : 0x00000000    R7 : 0x20005000
  R2 : 0x0000000C    R8 : 0x00000000
  R3 : 0x0000000C    R10: 0x00000000
  R4 : 0x00040034    R11: 0x00000000
  R5 : 0x20005000    R12: 0x20005750
  R9 : 0x20005800 (Static Base Register)
  SP : 0x200057D0 (Process Stack Pointer)
  LR : 0x00040D07
  PC : 0x00040D0E
 YPC : 0x00040D0E

 APSR: N 0 Z 0 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true 

 Total number of grant regions defined: 15
  Grant  0 : --          Grant  5 0x1: 0x20006a64  Grant 10 : --        
  Grant  1 : --          Grant  6 : --          Grant 11 0x60006: 0x20006a40
  Grant  2 0x0: 0x20006a28  Grant  7 : --          Grant 12 : --        
  Grant  3 : --          Grant  8 0x60004: 0x20006a50  Grant 13 : --        
  Grant  4 : --          Grant  9 0x60000: 0x20006a18  Grant 14 : --        

 Cortex-M MPU
  Region 0: [0x20005000:0x20006000], length: 4096 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20005000:0x20005200], Enabled
    Sub-region 1: [0x20005200:0x20005400], Enabled
    Sub-region 2: [0x20005400:0x20005600], Enabled
    Sub-region 3: [0x20005600:0x20005800], Enabled
    Sub-region 4: [0x20005800:0x20005A00], Enabled
    Sub-region 5: [0x20005A00:0x20005C00], Enabled
    Sub-region 6: [0x20005C00:0x20005E00], Enabled
    Sub-region 7: [0x20005E00:0x20006000], Enabled
  Region 1: [0x20006000:0x20007000], length: 4096 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20006000:0x20006200], Enabled
    Sub-region 1: [0x20006200:0x20006400], Disabled
    Sub-region 2: [0x20006400:0x20006600], Disabled
    Sub-region 3: [0x20006600:0x20006800], Disabled
    Sub-region 4: [0x20006800:0x20006A00], Disabled
    Sub-region 5: [0x20006A00:0x20006C00], Disabled
    Sub-region 6: [0x20006C00:0x20006E00], Disabled
    Sub-region 7: [0x20006E00:0x20007000], Disabled
  Region 2: [0x00040000:0x00044000], length: 16384 bytes; UnprivilegedReadOnly (0x2)
    Sub-region 0: [0x00040000:0x00040800], Enabled
    Sub-region 1: [0x00040800:0x00041000], Enabled
    Sub-region 2: [0x00041000:0x00041800], Enabled
    Sub-region 3: [0x00041800:0x00042000], Enabled
    Sub-region 4: [0x00042000:0x00042800], Enabled
    Sub-region 5: [0x00042800:0x00043000], Enabled
    Sub-region 6: [0x00043000:0x00043800], Enabled
    Sub-region 7: [0x00043800:0x00044000], Enabled
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug libtock-c apps, run `make debug RAM_START=0x20005000`
FLASH_INIT=0x4005d in the app's folder and open the .lst file.

𝐀𝐩𝐩: blink   -   [Yielded]
 Events Queued: 0   Syscall Count: 430342   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Yield { which: 1, address: 0x0 }
 Completion Code: None
 Credential: None

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20009000═╪══════════════════════════════════════════╝
             │ Grant Ptrs      120
             │ Upcalls         320
             │ Process         920
  0x20008AB0 ┼───────────────────────────────────────────
             │ ▼ Grant          24
  0x20008A98 ┼───────────────────────────────────────────
             │ Unused
  0x200079FC ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   4252               S
  0x200079FC ┼─────────────────────────────────────────── R
             │ Data            508 |    508               A
  0x20007800 ┼─────────────────────────────────────────── M
             │ ▼ Stack         216 |   2048          
  0x20007728 ┼───────────────────────────────────────────
             │ Unused
  0x20007000 ┴───────────────────────────────────────────
             .....
  0x00044800 ┬─────────────────────────────────────────── F
             │ App Flash      1996                        L
  0x00044034 ┼─────────────────────────────────────────── A
             │ Protected        52                        S
  0x00044000 ┴─────────────────────────────────────────── H

  R0 : 0x00000001    R6 : 0x000446A8
  R1 : 0x00000000    R7 : 0x20007000
  R2 : 0x00000009    R8 : 0x00000000
  R3 : 0x00000009    R10: 0x00000000
  R4 : 0x200077B3    R11: 0x00000000
  R5 : 0x00003639    R12: 0x20007788
  R9 : 0x20007800 (Static Base Register)
  SP : 0x20007780 (Process Stack Pointer)
  LR : 0x0004444F
  PC : 0x00044456
 YPC : 0x00044456

 APSR: N 0 Z 0 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true 

 Total number of grant regions defined: 15
  Grant  0 : --          Grant  5 : --          Grant 10 : --        
  Grant  1 : --          Grant  6 : --          Grant 11 : --        
  Grant  2 0x0: 0x20008a98  Grant  7 : --          Grant 12 : --        
  Grant  3 : --          Grant  8 : --          Grant 13 : --        
  Grant  4 : --          Grant  9 : --          Grant 14 : --        

 Cortex-M MPU
  Region 0: [0x20007000:0x20008000], length: 4096 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20007000:0x20007200], Enabled
    Sub-region 1: [0x20007200:0x20007400], Enabled
    Sub-region 2: [0x20007400:0x20007600], Enabled
    Sub-region 3: [0x20007600:0x20007800], Enabled
    Sub-region 4: [0x20007800:0x20007A00], Enabled
    Sub-region 5: [0x20007A00:0x20007C00], Disabled
    Sub-region 6: [0x20007C00:0x20007E00], Disabled
    Sub-region 7: [0x20007E00:0x20008000], Disabled
  Region 1: Unused
  Region 2: [0x00044000:0x00044800], length: 2048 bytes; UnprivilegedReadOnly (0x2)
    Sub-region 0: [0x00044000:0x00044100], Enabled
    Sub-region 1: [0x00044100:0x00044200], Enabled
    Sub-region 2: [0x00044200:0x00044300], Enabled
    Sub-region 3: [0x00044300:0x00044400], Enabled
    Sub-region 4: [0x00044400:0x00044500], Enabled
    Sub-region 5: [0x00044500:0x00044600], Enabled
    Sub-region 6: [0x00044600:0x00044700], Enabled
    Sub-region 7: [0x00044700:0x00044800], Enabled
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug libtock-c apps, run `make debug RAM_START=0x20007000`
FLASH_INIT=0x4405d in the app's folder and open the .lst file.

`

itp18506, 2023-06-10:
tock$ process music
𝐀𝐩𝐩: music   -   [Yielded]
 Events Queued: 0   Syscall Count: 200   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Yield { which: 1, address: 0x0 }
 Completion Code: None
 Credential: None

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20007000═╪════════�
panicked at 'kernel stack overflow', arch/cortex-m/src/lib.rs:636:13
	Kernel version release-2.1-1109-gf0501a03a

---| No debug queue found. You can set it with the DebugQueue component.

---| Cortex-M Fault Status |---
No Cortex-M faults detected.

---| App Status |---
𝐀𝐩𝐩: music   -   [Yielded]
 Events Queued: 0   Syscall Count: 200   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Yield { which: 1, address: 0x0 }
 Completion Code: None
 Credential: None

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20007000═╪══════════════════════════════════════════╝
             │ Grant Ptrs      120
             │ Upcalls         320
             │ Process         920
  0x20006AB0 ┼───────────────────────────────────────────
             │ ▼ Grant         100
  0x20006A4C ┼───────────────────────────────────────────
             │ Unused
  0x20006218 ┼───────────────────────────────────────────
             │ ▲ Heap         1512 |   3612               S
  0x20005C30 ┼─────────────────────────────────────────── R
             │ Data           1072 |   1072               A
  0x20005800 ┼─────────────────────────────────────────── M
             │ ▼ Stack         264 |   2048          
  0x200056F8 ┼───────────────────────────────────────────
             │ Unused
  0x20005000 ┴───────────────────────────────────────────
             .....
  0x00044000 ┬─────────────────────────────────────────── F
             │ App Flash     16332                        L
  0x00040034 ┼─────────────────────────────────────────── A
             │ Protected        52                        S
  0x00040000 ┴─────────────────────────────────────────── H

  R0 : 0x00000001    R6 : 0x00042078
  R1 : 0x00000000    R7 : 0x20005000
  R2 : 0x00000000    R8 : 0x00000000
  R3 : 0x00000000    R10: 0x00000000
  R4 : 0x00040034    R11: 0x00000000
  R5 : 0x20005000    R12: 0x99999999
  R9 : 0x20005800 (Static Base Register)
  SP : 0x200057D0 (Process Stack Pointer)
  LR : 0x000402D7
  PC : 0x000402DE
 YPC : 0x000402DE

 APSR: N 0 Z 0 C 1 V 0 Q 0
       GE 0 0 0 0
 EPSR: ICI.IT 0x00
       ThumbBit true 

 Total number of grant regions defined: 15
  Grant  0 : --          Grant  5 0x1: 0x20006a64  Grant 10 : --        
  Grant  1 : --          Grant  6 : --          Grant 11 : --        
  Grant  2 : --          Grant  7 : --          Grant 12 : --        
  Grant  3 0x90000: 0x20006a4c  Grant  8 : --          Grant 13 : --        
  Grant  4 : --          Grant  9 : --          Grant 14 : --        

 Cortex-M MPU
  Region 0: [0x20005000:0x20006000], length: 4096 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20005000:0x20005200], Enabled
    Sub-region 1: [0x20005200:0x20005400], Enabled
    Sub-region 2: [0x20005400:0x20005600], Enabled
    Sub-region 3: [0x20005600:0x20005800], Enabled
    Sub-region 4: [0x20005800:0x20005A00], Enabled
    Sub-region 5: [0x20005A00:0x20005C00], Enabled
    Sub-region 6: [0x20005C00:0x20005E00], Enabled
    Sub-region 7: [0x20005E00:0x20006000], Enabled
  Region 1: [0x20006000:0x20007000], length: 4096 bytes; ReadWrite (0x3)
    Sub-region 0: [0x20006000:0x20006200], Enabled
    Sub-region 1: [0x20006200:0x20006400], Enabled
    Sub-region 2: [0x20006400:0x20006600], Disabled
    Sub-region 3: [0x20006600:0x20006800], Disabled
    Sub-region 4: [0x20006800:0x20006A00], Disabled
    Sub-region 5: [0x20006A00:0x20006C00], Disabled
    Sub-region 6: [0x20006C00:0x20006E00], Disabled
    Sub-region 7: [0x20006E00:0x20007000], Disabled
  Region 2: [0x00040000:0x00044000], length: 16384 bytes; UnprivilegedReadOnly (0x2)
    Sub-region 0: [0x00040000:0x00040800], Enabled
    Sub-region 1: [0x00040800:0x00041000], Enabled
    Sub-region 2: [0x00041000:0x00041800], Enabled
    Sub-region 3: [0x00041800:0x00042000], Enabled
    Sub-region 4: [0x00042000:0x00042800], Enabled
    Sub-region 5: [0x00042800:0x00043000], Enabled
    Sub-region 6: [0x00043000:0x00043800], Enabled
    Sub-region 7: [0x00043800:0x00044000], Enabled
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug libtock-c apps, run `make debug RAM_START=0x20005000`
FLASH_INIT=0x4005d in the app's folder and open the .lst file.

itp18506, 2023-06-10:
increasing the stack buffer from board's main.rs file fixed the issue.

_#[link_section = ".stack_buffer"]
// pub static mut STACK_MEMORY: [u8; 0x1100] = [0; 0x1100];
// debug mode requires more stack space
pub static mut STACK_MEMORY: [u8; 0x2000] = [0; 0x2000];_

lschuermann, 2023-06-10:
@itp18506 Thanks for bringing this to our attention. Did you enable "debug mode" (I presume this refers to the process loading printing, etc.), or did you encounter this with the regular board configuration? If so, we'd appreciate a pull request, or just a comment here, such that we can fix this for other users as well.

bradjc, 2023-06-10:
We should look into if that command in process console uses an expected
amount of stack or if we can optimize it.

hudson-ayers, 2023-06-25:
Closing this because I think there are PRs open to address this from both angles.

closed 2023-06-25 by hudson-ayers