tock/tock #3484: How does VirtualHmac handle clients?

bradjc, 2023-06-14:
I'm looking at virtual_hmac.rs and the only set_client() I see is here:

https://github.com/tock/tock/blob/829cde6e32bcefcfce1d8a9bb5a157512cbba134/capsules/core/src/virtualizers/virtual_hmac.rs#L176-L182

which sets the client in the underlying HMAC driver. Wouldn't that mean that only one client (the last) can exist?

The virtual object has a place for a client:

https://github.com/tock/tock/blob/829cde6e32bcefcfce1d8a9bb5a157512cbba134/capsules/core/src/virtualizers/virtual_hmac.rs#L21-L31

but while it is read from I don't think it is ever set.

Am I missing something?

alistair23, 2023-06-15:
It could be a bug. I'm don't think there are currently multiple users of HMAC so it probably isn't well tested

bradjc, 2023-06-15:
I don't know how to fix this because the traits get in the way:

```
❯ make allcheck
    Checking capsules-core v0.1.0 (/Users/bradjc/git/tock/capsules/core)
   Compiling redboard_artemis_nano v0.1.0 (/Users/bradjc/git/tock/boards/apollo3/redboard_artemis_nano)
   Compiling particle_boron v0.1.0 (/Users/bradjc/git/tock/boards/particle_boron)
   Compiling lora_things_plus v0.1.0 (/Users/bradjc/git/tock/boards/apollo3/lora_things_plus)
   Compiling sma_q3 v0.1.0 (/Users/bradjc/git/tock/boards/sma_q3)
error[E0277]: the trait bound `A: HmacSha256` is not satisfied
   --> capsules/core/src/virtualizers/virtual_hmac.rs:75:38
    |
75  |             self.mux.hmac.set_client(self);
    |                                      ^^^^ the trait `HmacSha256` is not implemented for `A`
    |
note: required for `VirtualMuxHmac<'a, A, L>` to implement `ClientData<L>`
   --> capsules/core/src/virtualizers/virtual_hmac.rs:193:7
    |
191 |         A: digest::Digest<'a, L> + digest::HmacSha256 + digest::HmacSha384 + di...
    |                                    ------------------ unsatisfied trait bound introduced here
192 |         const L: usize,
193 |     > digest::ClientData<L> for VirtualMuxHmac<'a, A, L>
    |       ^^^^^^^^^^^^^^^^^^^^^     ^^^^^^^^^^^^^^^^^^^^^^^^
    = note: required for `VirtualMuxHmac<'a, A, L>` to implement `kernel::hil::digest::Client<L>`
    = note: required for the cast from `VirtualMuxHmac<'a, A, L>` to the object type `dyn kernel::hil::digest::Client<L>`
help: consider further restricting this bound
    |
63  | impl<'a, A: digest::Digest<'a, L> + kernel::hil::digest::HmacSha256, const L: usize> digest::DigestData<'a, L>
    |                                   +++++++++++++++++++++++++++++++++

error[E0277]: the trait bound `A: HmacSha384` is not satisfied
   --> capsules/core/src/virtualizers/virtual_hmac.rs:75:38
    |
75  |             self.mux.hmac.set_client(self);
    |                                      ^^^^ the trait `HmacSha384` is not implemented for `A`
    |
note: required for `VirtualMuxHmac<'a, A, L>` to implement `ClientData<L>`
   --> capsules/core/src/virtualizers/virtual_hmac.rs:193:7
    |
191 |         A: digest::Digest<'a, L> + digest::HmacSha256 + digest::HmacSha384 + di...
    |                                                         ------------------ unsatisfied trait bound introduced here
192 |         const L: usize,
193 |     > digest::ClientData<L> for VirtualMuxHmac<'a, A, L>
    |       ^^^^^^^^^^^^^^^^^^^^^     ^^^^^^^^^^^^^^^^^^^^^^^^
    = note: required for `VirtualMuxHmac<'a, A, L>` to implement `kernel::hil::digest::Client<L>`
    = note: required for the cast from `VirtualMuxHmac<'a, A, L>` to the object type `dyn kernel::hil::digest::Client<L>`
help: consider further restricting this bound
    |
63  | impl<'a, A: digest::Digest<'a, L> + kernel::hil::digest::HmacSha384, const L: usize> digest::DigestData<'a, L>
    |                                   +++++++++++++++++++++++++++++++++

error[E0277]: the trait bound `A: HmacSha512` is not satisfied
   --> capsules/core/src/virtualizers/virtual_hmac.rs:75:38
    |
75  |             self.mux.hmac.set_client(self);
    |                                      ^^^^ the trait `HmacSha512` is not implemented for `A`
    |
note: required for `VirtualMuxHmac<'a, A, L>` to implement `ClientData<L>`
   --> capsules/core/src/virtualizers/virtual_hmac.rs:193:7
    |
191 |         A: digest::Digest<'a, L> + digest::HmacSha256 + digest::HmacSha384 + digest::HmacSha5...
    |                                                                              ------------------ unsatisfied trait bound introduced here
192 |         const L: usize,
193 |     > digest::ClientData<L> for VirtualMuxHmac<'a, A, L>
    |       ^^^^^^^^^^^^^^^^^^^^^     ^^^^^^^^^^^^^^^^^^^^^^^^
    = note: required for `VirtualMuxHmac<'a, A, L>` to implement `kernel::hil::digest::Client<L>`
    = note: required for the cast from `VirtualMuxHmac<'a, A, L>` to the object type `dyn kernel::hil::digest::Client<L>`
help: consider further restricting this bound
    |
63  | impl<'a, A: digest::Digest<'a, L> + kernel::hil::digest::HmacSha512, const L: usize> digest::DigestData<'a, L>
    |                                   +++++++++++++++++++++++++++++++++
```

Work in progress here: https://github.com/tock/tock/compare/virtual-hmac-clients?expand=1

Anyone know how to fix this? Otherwise we need to remove these virtual capsules or rethink the HIL.

bradjc, 2023-06-15:
Also, how does queuing setting the mode work? Specifically, the `else` here:

https://github.com/tock/tock/blob/829cde6e32bcefcfce1d8a9bb5a157512cbba134/capsules/core/src/virtualizers/virtual_hmac.rs#L243-L255

The operation is queued, but setting the mode does not generate a callback. So the caller receives an `Ok(())`, and proceeds to the next task, probably adding data to the hmac engine. Adding data _does_ generate a callback, so it waits. When the virtualizer gets to handle the set mode request, it does so, then returns. And it seems like everything stops there.

hudson-ayers, 2023-07-10:
Closed by #3503 , once it merges.

closed 2023-07-17 by ppannuto