tock/tock #3495: `kv_system` HIL: why is hashed the key the full buffer?

bradjc, 2023-06-16:
In the KV HIL, the `generate_key()` function takes in a buffer for the key without a length:

https://github.com/tock/tock/blob/b28fcfb601846ee0d3e5a4605e4c5f149d5cbdf5/kernel/src/hil/kv_system.rs#L165-L176

In practice this makes the resulting hash dependent on the size of the static buffer that the board happened to use.

However, in our implementation of the kv_system HIL, the buffer is converted to a leasable buffer before being passed to the hasher:

https://github.com/tock/tock/blob/b28fcfb601846ee0d3e5a4605e4c5f149d5cbdf5/capsules/extra/src/tickv.rs#L395-L417

Is there a reason to not use `LeasableMutableBuffer` in the kv_system trait?

alistair23, 2023-06-18:
I don't think there is a reason, we probably should use `LeasableMutableBuffer`

bradjc, 2023-07-21:
This has been fixed.

closed 2023-07-21 by bradjc