tock/tock #3497: EarlGrey CW310 QEMU `spi_host` test bug

lschuermann, 2023-06-19:
Over the past few weeks, I've noticed the `spi_host` QEMU CI test fail a couple of times, on PRs which are entirely unrelated to that board / subsystem. We should investigate whether there's a race either in the Tock drivers for that board, or in QEMU. Example of failing workflow: https://github.com/tock/tock/actions/runs/5312037775/jobs/9616026681

alistair23, 2023-06-20:
@twilfredo

alistair23, 2023-06-20:
In case the logs get deleted:

```
[SPI] Setup spi_host0 partial_transfer... 

panicked at 'assertion failed: `(left == right)`
  left: `false`,
 right: `true`', boards/opentitan/earlgrey-cw310/src/tests/spi_host.rs:163:5
	Kernel version aab0546

---| No debug queue found. You can set it with the DebugQueue component.

---| OpenTitan Earlgrey configuration for fpga_cw310 |---
---| RISC-V Machine State |---
Last cause (mcause): Machine external interrupt (interrupt=1, exception code=0x0000000B)
Last value (mtval):  0x00000000

System register dump:
 mepc:    0x20021650    mstatus:     0x00000088
 mcycle:  0x2194CE6E    minstret:    0x219505E2
 mtvec:   0x20000501
 mstatus: 0x00000088
  uie:    false  upie:   false
  sie:    false  spie:   false
  mie:    true   mpie:   true
  spp:    false
 mie:   0x00000808   mip:   0x00000000
  usoft:  false               false 
  ssoft:  false               false 
  msoft:  true                false 
  utimer: false               false 
  stimer: false               false 
  mtimer: false               false 
  uext:   false               false 
  sext:   false               false 
  mext:   true                false 
 ePMP regions:
  [0]: addr=0x00000000, end=0x00000000, cfg=0x0 (OFF) (----)
  [1]: addr=0x00000000, end=0x00000000, cfg=0x0 (OFF) (----)
  [2]: addr=0x00000000, end=0x00000000, cfg=0x0 (OFF) (----)
  [3]: addr=0x00000000, end=0x00000000, cfg=0x0 (OFF) (----)
  [4]: addr=0x00000000, end=0x00000000, cfg=0x0 (OFF) (----)
  [5]: addr=0x00000000, end=0x00000000, cfg=0x0 (OFF) (----)
  [6]: addr=0x00000000, end=0x40000000, cfg=0x0 (OFF) (----)
  [7]: addr=0x40000000, end=0x49000000, cfg=0x8B (TOR) (lrw-)
  [8]: addr=0x49000000, end=0x10008474, cfg=0x0 (OFF) (----)
  [9]: addr=0x10008474, end=0x10010000, cfg=0x8B (TOR) (lrw-)
  [10]: addr=0x10010000, end=0x20060000, cfg=0x0 (OFF) (----)
  [11]: addr=0x20060000, end=0x20090000, cfg=0x8B (TOR) (lrw-)
  [12]: addr=0x20090000, end=0x20000000, cfg=0x0 (OFF) (----)
  [13]: addr=0x20000000, end=0x2002EB54, cfg=0x8D (TOR) (lr-x)
  [14]: addr=0x2002EB54, end=0x10000650, cfg=0x0 (OFF) (----)
  [15]: addr=0x10000650, end=0x10008474, cfg=0x8B (TOR) (lrw-)

---| App Status |---
panicked at 'assertion failed: `(left == right)`
  left: `false`,
 right: `true`', boards/opentitan/earlgrey-cw310/src/tests/spi_host.rs:163:5
error: test failed, to rerun pass `-p earlgrey-cw310 --bin earlgrey-cw310`
```

twilfredo, 2023-06-21:
I tested this locally on the latest qemu commit, and I'm not seeing any errors see log: 

```
qemu-system-riscv32: warning: disabling zca extension for hart 0x00000000 because privilege spec version does not match
Failed to set flash memory protection: NOSUPPORT
Unable to find otbn-rsa, disabling RSA support
OpenTitan initialisation complete. Entering main loop
check run AES128 CBC... 
    [ok]
check run AES128 CCM... 
AES CCM* encryption/decryption tests
virtual_aes_ccm: no ccm_client is registered in VirtualAES128CCM
virtual_aes_ccm: no buffer is binded with VirtualAES128CCM
    [ok]
check run AES128 CTR... 
    [ok]
check run AES128 ECB... 
    [ok]
check run AES128 GCM... 
    [ok]
check run CSRNG Entropy 32... 
    [ok]
[FLASH_CTRL] Test page erase....
    [ok]
[FLASH_CTRL] Test memory protection api....
    [ok]
[FLASH_CTRL] Test memory protection functionality....
    [ok]
[FLASH_CTRL] Test page read/write....
    [ok]
check hmac load binary... 
    [ok]
check hmac check verify... 
    [ok]
start multi alarm test...
    [ok]
check otbn run binary...
    [FAIL] No OTBN binary
Not running RSA tests
Not running RSA tests
check rsa key import... 
    [ok]
Not running RSA tests
check rsa 4096 bit key import... 
    [ok]
start SHA256 verify test
Sha256Test: Setting slice to 12..24
Sha256Test: Setting slice to 24..36
Sha256Test: Setting slice to 36..48
Sha256Test: Setting slice to 48..60
Sha256Test: Setting slice to 60..72
Sha256Test: Verification result: Ok(true)
    [ok]
check SipHash 2-4... 
    [ok]
[SPI] Setup spi_host0 partial_transfer... 
    [ok]
[SPI] Setup spi_host0 transfers... 
    [ok]
[SPI] Setup spi_host0 transfer...x2 
    [ok]
start TicKV append key test...
---Starting TicKV Tests---
Generated key: [61, 245, 130, 62, 163, 73, 69, 239]
Now appending the key
    [ok]
trivial assertion... 
    [ok]
```
Should we just bump the qemu commit and see what happens with CI?

lschuermann, 2023-06-21:
From what I observed, this seems to be a transient issue / race condition. It only fails on a number of CI runs. I don't know how to best debug this, but haven't looked at the exact issue yet -- maybe it's something obvious in the respective Tock / QEMU drivers?

Otherwise it generally seems good to keep things up to date. If it goes away then, I suppose that's good enough? Would be hard to be sure though.

twilfredo, 2023-06-22:
Yeah it does seem that way and tricky to debug, I've sent this for now https://github.com/tock/tock/pull/3501 hoping it just works...

About where it is failing in the tests, it does seems like an interrupt didn't land in the very first test for `spi_host` OR somehow QEMU didn't transfer all the bytes in the buffer  (seems unlikely).

alistair23, 2023-06-22:
If the issues only happens occasionally in the CI then a good way to test would be to load up your machine (run a big build) and then run the tests in a loop. See if you can reproduce the failure. It looks like an interrupt isn't being received.

The other option would be to enable more logging from QEMU with the CI (add a `-d int` at least to the command line arguments) and see if that helps narrow it down when the failure happens.

Adding a `-d int` to the QEMU command line is probably the easiest first step, then we can just monitor from there

closed 2023-06-23 by bradjc