tock/tock #3518: Tracking: Use `start()` function in board main.rs

bradjc, 2023-06-27:
Since the main() function never returns, its stack frame is never reclaimed. If that stack frame is large, that results in significant stack memory waste.

We tried addressing this in #2425. However, in retrospect that was a piecemeal approach, where at the time peripheral creation was the only thing causing main's stack frame to be large. Since then, we've realized other operations cause a large stack frame (for example inlining the load_processes function #3499).

We should just avoid this once-and-for-all by moving everything that is currently in `main()` into a new function called `start()`, except for the call to `kernel::kernel_loop()`.

Boards to be updated:
- [ ] microbit_v2: #3514 
- [ ] all others

twilfredo, 2023-11-07:
The Particle Boron now has support for start() fixed in https://github.com/tock/tock/pull/3733

bradjc, 2024-07-12:
Done

closed 2024-07-12 by bradjc