tock/tock #3529: Migrate from legacy mangling scheme to v0 scheme

nbdd0121, 2023-06-29:
Tock currently uses the default symbol mangling scheme, which is the legacy scheme at the moment, but will eventually be moved to the v0 mangling scheme.

The v0 scheme contains more information (most notably, generic arguments), e.g. this is the demangling in v0:
```rust
<<capsules_core::led::LedDriver<kernel::hil::led::LedHigh<lowrisc::gpio::GpioPin>, 8> as kernel::syscall_driver::SyscallDriver>::allocate_grant>
```
where this is for legacy:
```rust
<<capsules_core::led::LedDriver<L,_> as kernel::syscall_driver::SyscallDriver>::allocate_grant>
```

There are a few scripts that currently rely on the legacy scheme, e.g. `print_tock_memory_usage.py` and `find_panics.py` which would need to be migrated to support the v0 scheme. Some scripts, e.g. `stack_analysis.sh` on the other rely on v0 scheme.

We should migrate entirely away from legacy scheme to the v0 mangling scheme.

Rust tracking issue for the v0 mangling scheme: https://github.com/rust-lang/rust/issues/60705

nbdd0121, 2023-06-29:
I can work on the migration once it's acked by maintainers to be desirable.

lschuermann, 2023-06-29:
I don't see a reason why we wouldn't want this, except for the fact that it appears to be adding another (implicit) dependency on nightly Rust. We've been getting wary about these in the recent past. I have a feeling that this is likely to land before some of the other features we currently have, but maybe you should wait for some others to chime in.

For reference:
```
$ git grep -h "#\!\[feature" | sort | uniq
#![feature(alloc)]
#![feature(asm_const)]
#![feature(core_intrinsics)]
#![feature(custom_test_frameworks)]
#![feature(naked_functions)]
```

PR #3539 (2023-07-18): Switch to v0 mangling scheme and port print_tock_memory_usage.py
### Pull Request Overview

This PR switches the symbol mangling scheme to v0. I also ported the `print_tock_memory_usage.py` script to parse the demangled symbol from v0 scheme.

Fix #3529

### Testing Strategy

`make -C boards/opentitan/earlgrey-cw310/ memory`

### TODO or Help Wanted

`tools/find_panics.py` currently does not work

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2023-07-18 by bradjc