tock/tock #3575: Tock Extension for VSCode

bradjc, 2023-07-28:
Many embedded developers are familiar with using IDEs for development. VSCode is quite popular. Creating a Tock extension for VSCode that setup a dev environment with the relevant source repos and tools would help new developers to Tock and those following the tutorials.

- https://code.visualstudio.com/api
- https://github.com/microsoft/vscode-extension-samples/tree/main/custom-data-sample

ppannuto, 2023-07-28:
Is this a case of updating this: https://github.com/tock/tock/tree/master/.vscode, or building something new / different?

bradjc, 2023-07-31:
> Is this a case of updating this: https://github.com/tock/tock/tree/master/.vscode, or building something new / different?

I think this is something different. The goal would be to have this be installable through vscode, and include "everything", ie a tock checkout, libtock checkout, tockloader, etc.

twilfredo, 2023-09-26:
> I think this is something different. The goal would be to have this be installable through vscode, and include "everything", ie a tock checkout, libtock checkout, tockloader, etc.

I've seen something similar with other OS's using [PlatformIO](https://platformio.org/) (which has a VSCode extension). For instance [Zephyr](https://docs.platformio.org/en/stable/frameworks/zephyr.html). It's able to setup the build environment and although it seems to be directed more at the user level for writing apps and flashing.

Was that what you had in mind? Or was this particularly aimed towards development.

bradjc, 2023-09-28:
> I've seen something similar with other OS's using [PlatformIO](https://platformio.org/) (which has a VSCode extension). For instance [...]

Short answer: yes. Minimal friction to get people developing with Tock, whether userspace or kernel, ideally both.