tock/tock #3581: Tock Demo Apps

bradjc, 2023-07-31:
At TockWorld 6, two separate discussions arrived on the need for Tock to maintain a handful of "showcase applications" that use the Tock kernel to implement tangible, realistic applications. These would give new users a starting place to explore Tock, as well as be good demos to show off Tock's capabilities.

This issue is a place to discuss ideas as well as catalog "showcase applications".

- [HOTP](https://github.com/tock/libtock-c/tree/master/examples/tutorials/hotp)

bradjc, 2023-09-05:
One idea is a weather station based on:

- https://www.sparkfun.com/products/22636
- https://www.sparkfun.com/products/16984

bradjc, 2024-02-12:
I'm planning to write a soil moisture tutorial with:

- one app reading analog sensor
- one app sending the data using thread
- one app deciding when to water the plant
- app 1 and 3 with access to a screen to show the moisture level and whether the plan needs to be watered

alistair23, 2024-02-13:
You can also send the data using LoRa, which should provide some more range as well

bradjc, 2024-02-13:
> You can also send the data using LoRa, which should provide some more range as well

I don't currently have a Tock-with-LoRa board :(

viswajith-g, 2025-05-05:
At this point in time we are able to send data with LoRa, so that should be updated. But we can also add

1. Dynamic Apps
2. Process Info 

Or the combined tutorial. ProcessInfo acts like a process inspector analogous to the task manager on Windows so that is an easy comparison for people to draw.

lschuermann, 2025-05-06:
@bradjc Do we have a document that we can incorporate these ideas into? I wouldn't want them to get lost in the void of this issue tracker.

alistair23, 2025-05-07:
There is a Tock based weather station write up available: https://www.alistair23.me/2025/03/15/tock-lorawan-weather-station/