tock/tock #3612: ESP32 wrong register offsets

CosminGGeorgescu, 2023-08-12:
The first occurrence I've found of this can be found [here](https://github.com/tock/tock/blob/1a111a3e748815117b0ef939ab730b31d77a409d/chips/esp32/src/rtc_cntl.rs#L29). There is no "timer 6" found in the official [esp32](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#page=691) doc, nor do the offsets match with any of the [esp32s2](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf#page=280), [esp32c3](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#page=220) or [esp32s3](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf#page=567) docs. All of the registers in the range 0x30 - 0x68 have their offset mapped with a +0x4 added to their actual offset.

Moreover, the `@END` marker offset does not match with any of the aforementioned chip's LowPower Management actual address space. File in question is included in the `esp32` package and used as an import for the `esp32c3` [struct](https://github.com/tock/tock/blob/1a111a3e748815117b0ef939ab730b31d77a409d/chips/esp32-c3/src/chip.rs#L40), which has a 32bit [register](https://www.espressif.com/sites/default/files/documentation/esp32-c3_technical_reference_manual_en.pdf#page=258&zoom=100,76,65) starting at the `@END` offset. The ends differ greatly between chips, but these are the only ones we seem to have a package for right now.

While, technically, the implementation remains correct, as neither of the wrongly mapped registers are used (currently?), it is theoretically wrong. I haven't checked side-by-side the `esp32` and `esp32c3` registers so I don't know if these are the only differences. I would suggest declaring a new `rtc_cntl.rs` file that resides in the `esp32c3` and copying the current implementation plus the new registers, but it is ultimately your call. I'll tag @alistair23👋 as Git tells me he saw to the current file being used.

gobftald, 2023-08-13:
I’m writing/porting a tock wifi driver for the esp32-c3-devkitM-1 board. I’ve also bumped into your problem, even I needed other — not documented — peripheral drivers like “apb_ctrl" for wifi. My best and most precise source of information for "chips drivers" are Espressif’s official crates. Although their “hal team" use another method for defining “[peripheral](https://github.com/esp-rs/esp-hal/blob/main/esp-hal-common/src/peripheral.rs)” but you can find the register definition and address for all esp32xx chips in their specific esp-rs/esp-pacs (Peripheral Access Crates), generated from the factory [svd](https://github.com/esp-rs/esp-pacs/tree/main/esp32c3/svd) file. So it should be the most authentic definition for all peripherals, also for those which are not documented yet in the Technical Reference Manuals. So in this repository you can easily find [esp32c3 rtc_cntl.rs](https://github.com/esp-rs/esp-pacs/blob/main/esp32c3/src/rtc_cntl.rs). Its bitfield definitions can also be found [there](https://github.com/esp-rs/esp-pacs/tree/main/esp32c3/src/rtc_cntl). The generated files naturally don’t follow our (tock) register and bitfield definition macros but based on the "doc comments” they are quite trivial. Based on them I was easily able to write my own tock "chips drivers".

bradjc, 2023-09-05:
Is there a todo here or should we close this?

gobftald, 2023-09-05:
I think, based on the published factory crates, it is solved.

CosminGGeorgescu, 2023-09-05:
Can be closed, just needed some insight. Thank you @gobftald and @alistair23 !

closed 2023-09-05 by CosminGGeorgescu