tock/tock #3650: Can not get negative temperature via TemperatureSensor

tonny1983, 2023-08-30:
Let's start the problem from the latest release version 2.1.1, in which the definition of `TemperatureClient` trait is:
```rust
pub trait TemperatureClient {
    fn callback(&self, value: usize);
}
```

Obviously, the `value` there is unsigned, and therefore, negative value always  becomes 0.

Now, in the master brench, the definition of `TemperatureClient` trait seems to be changed for the problem:
```rust
pub trait TemperatureClient {
    fn callback(&self, value: Result<i32, ErrorCode>);
}
```
However, the `TemperatureSensor` which implements the `TemperatureClient` trait casts the `i32` type to `usize` with the following code:
```rust
upcalls.schedule_upcall(0, (temp_val as usize, 0, 0)).ok();
```

May I know the reason why the temperature should be `usize`? And how I can get negative temperature now?

ppannuto, 2023-08-30:
At the kernel/userspace boundary, there is no real type information, just numbers in registers ([ABI docs](https://github.com/tock/tock/blob/master/doc/reference/trd104-syscalls.md#31-registers)).

The cast here doesn't change the value, it just coerces to the number to the type expected by the syscall interface. Userspace [will get that same number](https://github.com/tock/tock/blob/master/doc/syscalls/60000_ambient_temperature.md#subscribe-number-0), and can just cast back to the `i32` or `int32_t` or whatever is appropriate for the runtime in use.

tonny1983, 2023-08-31:
Thanks a lot for your reply. I think I found the difference between version 2.1.1 and the master branch that the conversion from `f32` to `usize` (in 2.1.1) will lost value, but on the other side, the conversion from `f32` to `usize` via `i32` (in the master branch) will keep the value.

closed 2023-08-31 by tonny1983