tock/tock #3659: RP2040 GPIO Possible Bug

brghena, 2023-09-07:
In the `handle_interrupt()` function in the `gpio` implementation for the RP2040 there are two possible bugs.

The first is seen in a lack of parallelism between the branches of the if/else chain. If `(1 << l_low_reg_no + 2) != 0`, then `self.gpio_registers.intr[bank_no].set()` is called. This happens for "+ 3" as well, but not for "+1" (or "+0"). This might be intended behavior, but certainly sticks out as strange.

The second is a question about intended order of operations. `(1 << l_low_reg_no + 2)` is actually `(1 << (l_low_reg_no + 2))`. Is this intended? I believe the answer is yes, but I'm not certain. 

Relevant lines of code are here: https://github.com/tock/tock/blame/master/chips/rp2040/src/gpio.rs#L342-L362

After the clippy changes from #3637 these lines become: https://github.com/tock/tock/blob/f3e310b26cb3775f0cbc91a9b9fd6bc14c4e8ab8/chips/rp2040/src/gpio.rs#L342-L362

@alexandruradovici @ioanaculic @irina-b-dev could any of you confirm or reject this bug?

alexandruradovici, 2023-09-22:
> In the `handle_interrupt()` function in the `gpio` implementation for the RP2040 there are two possible bugs.
>
> The first is seen in a lack of parallelism between the branches of the if/else chain. [...]
This is correct, the INTR register has packs of four bits, each of them representing an interrupt trigger. Level interrupts, represented at offsets 0 and 1, are cannot be cleared while edge interrupts, represented by offset 2 and 3, are `write-clear`.

[image]

> The second is a question about intended order of operations. `(1 << l_low_reg_no + 2)` is actually `(1 << (l_low_reg_no + 2))`. Is this intended? I believe the answer is yes, but I'm not certain.
> 
This was indeed a bug and is seems to be fixed now.

brghena, 2023-09-22:
@alexandruradovici So I think the takeaway here is that there are no bugs here and this was a false positive?

alexandruradovici, 2023-09-24:
I think so.

closed 2023-09-24 by brghena