tock/tock #3670: Derive Debug on tock_registers Value

andreeaflorescu, 2023-09-15:
Is there any reason for not deriving `Debug` on `Value`? I think it would be nice to have debug so that we can use `unwrap` on `read_as_enum`, which would simplify the code. If there is no technical reason for not having Debug, I can work on a patch to add this.

bradjc, 2023-09-18:
Seems fine to me.

lschuermann, 2023-09-22:
Apologies for my delayed response! We had this come up in the past, and IIRC the line of argument as to why we didn't implement `Debug` was that this is an easy way to accidentally generate a lot of code that increases flash size (e.g., when you print a register with `debug!`).

However, that's not a particularly good argument, given that there are many other ways to increase flash utilization accidentally. So I think this is a good change.

PR #3687 (2023-09-25): derive Debug on Value
### Pull Request Overview

This helps mostly with testing for doing assert_eq instead of comparing using a match on values.
Fixes: #3670 

### Testing Strategy

I ran the unit test. I've updated the documentation example of `read_as_enum` to use assert_eq directly rather than using a match to compare the values, thus validating that Debug is properly implemented.

### TODO or Help Wanted

N/A

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2023-09-25 by bradjc