tock/tock #3720: Second UART for nRF52840 (and nRF52833)

Niederb, 2023-10-24:
According to the datasheet the nRF52840 (and nRF52833) chips have a second UART available.
https://infocenter.nordicsemi.com/topic/struct_nrf52/struct/nrf52.html

Could the second UART be activated and used by adding a `uart::Uarte` field to `Nrf52840DefaultPeripherals` and adjusting the base address? The base address would be 0x40028000. 
Should this work or am I missing something (Clocks, interrupts, ...)?

lschuermann, 2023-10-24:
Indeed your proposed approach would work. Most peripheral drivers in Tock take a reference to their respective register file (`StaticRef`), which would all multiple instantiations of the same driver over different hardware instances. This does not seem to be the case for the nRF52 UART driver right now, but we should definitely be open to changing that. Here's an example of what that may look like:

https://github.com/tock/tock/blob/69f5bf8d2b48dd3af56477ef1d6eeec74285244d/chips/qemu_rv32_virt_chip/src/uart.rs#L222-L225

Note that the `DefaultPeripherals` infrastructure has been explicitly designed such that you can instantiate a custom set of peripherals in your own board definition. Once the UART peripheral has been modified to take a register reference in its constructor, you can instantiate your own such peripheral set in a custom board module, by defining a custom peripherals struct similar to the `Nrf52840DefaultPeripherals`. The upstream `Nrf52840DefaultPeripherals` is supposed to cover most commonly used peripherals, and peripherals exposed on development boards. Thus we may not want to instantiate multiple UARTs in that particular peripheral set.

Niederb, 2023-10-25:
Thanks for the quick response. I implemented the changes to make the registers configurable in #3722.
I hope this is what you had in mind.

Niederb, 2023-11-02:
Since #3722 has been merged I close this issue.

closed 2023-11-02 by Niederb