tock/tock #3756: Add `WriteableProcessBuffer::mut_ptr`

kupiakos, 2023-12-09:
`<ReadWriteProcessBuffer as ReadableProcessBuffer>::ptr` returns `*const u8`, and due to its implementation, it's valid to `.cast_mut()` that and then write to it so long as it's not empty (and null). A `WriteableProcessBuffer::mut_ptr` function would be much less risky against implementation change.

kupiakos, 2023-12-11:
An alternative that would not be a breaking change: document explicitly that for any `impl WriteableProcessBuffer`, the return value of `<ReadWriteProcessBuffer as ReadableProcessBuffer>::ptr` must be a writeable pointer. This should be indicated in the `unsafe trait` requirements.

lschuermann, 2024-11-10:
> `<ReadWriteProcessBuffer as ReadableProcessBuffer>::ptr` returns `*const u8` and, due to its implementation, it's valid to `.cast_mut()` that and then (in a single thread) write to it so long as it's not empty (and null). A `WriteableProcessBuffer::mut_ptr` function would be much less risky against implementation change and rely on more explicit promises.

This makes sense to me, and I'd appreciate a PR introducing this function.

However, this is not entirely accurate. Because this pointer is obtained pre-`enter` (and potentially as a way to circumvent `enter`), this pointer is only valid for so long as this is still a valid allowed buffer from a process that still "exist" (to mirror the terminology around `process_map_or`, which `enter` uses internally as a process liveness check).

A PR that introduces the `ptr_mut` function should thus also add some safety documentation that encompasses this condition.

lschuermann, 2024-11-10:
Let's make sure to also backport #4231 to `WriteableProcessBuffer` in that PR.