tock/tock #3797: kernel: uart: configurable rx buffer sizes at compile time (?)

twilfredo, 2024-01-15:
## Issue

For apps that rely on UART  for data transfers with RX frames greater than the RX buffer sizes specified by the kernel in [1] and [2], calling the receive userspace api (testing with libtock-rs) will error out (as expected).

Could we make these parameters (`RX_BUF_LEN` and `DEFAULT_BUF_SIZE`) configurable at compile time, perhaps by _somehow_ setting it at the board level or as an additional build parameter? Is there a more Tock idiomatic approach for this?

An example app that would benefit from something like this is: https://github.com/tock/libtock-rs/pull/525 , this is a sample app that allows for MCTP communications between two devices over USB to I2C. Using the following topology;

`[Host PC] <-- UART --> [Usb-I2C Bridge Device] <-- I2C --> [Target Endpoint]`

[1] https://github.com/tock/tock/blob/790a88412951fda1059747efe1792c748c5cad54/capsules/core/src/console.rs#L57
[2] https://github.com/tock/tock/blob/790a88412951fda1059747efe1792c748c5cad54/capsules/core/src/virtualizers/virtual_uart.rs#L57

bradjc, 2024-01-16:
Those constants should be thought of as just defaults. Boards can absolutely use different values. The primary way to do this is to pass the buffer length to the `_component_static!()` macro. Although very few components currently support this as we just haven't needed it very much.

twilfredo, 2024-01-16:
> Those constants should be thought of as just defaults. Boards can absolutely use different values. The primary way to do this is to pass the buffer length to the `_component_static!()` macro. Although very few components currently support this as we just haven't needed it very much.

ah great, thanks!

twilfredo, 2024-01-19:
fixed!

closed 2024-01-19 by twilfredo