tock/tock #3854: Kernel Unit Testing

JADarius, 2024-02-16:
I am in the process of writing unit tests for the kernel, but the module I'm writing for is dependent on external crates (for example the `tock_tbf` crate). My idea is to create some *fake* dependencies in the `tests` module and use conditional compiling in the struct declaration. For example, to emulate `tock_tbf::types::TbfHeader` inside I would rewrite the definition of `ProcessStandard` like so:
```
    #[cfg(not(test))]
    header: tock_tbf::types::TbfHeader,

    #[cfg(test)]
    header: tests::TbfHeader,
```
then I would create the `TbfHeader` struct inside the `tests` module and write the functions to give me the desired outputs.

I am not sure if this is the best way of writing unit tests for Tock and I would really like some feedback about this.

lschuermann, 2024-02-16:
From today's call, where we talked about this issue. An alternative approach to this could be to use traits and generics for the "real" and "mock" implementations of the different subsystems respectively. A concern that was voiced is that this generic argument would then need to be carried throughout much of Tock's codebase. A benefit would be that we would have a shared explicit interface between both implementations, which may reduce the potential for mismatches to evolve between the two implementations.

If the fact that this generic argument would need to be carried through much of Tock's codebase is the primary concern, we may be able to alleviate that by assigning a default type to that generic.

bradjc, 2024-02-16:
I don't think we should make Tock code harder to read and harder to write for unit testing. That feels like the wrong-way-round. We want to build the best OS to run, not to test.

If we need a stub library, like the TBF headers, that seems like a cargo.toml issue. Can't we just compile the kernel crate with a different dependency? Even if that requires some extra scripts to accomplish, that seems fine, and overall a _bunch_ less work than propagating a generic argument everywhere.

JADarius, 2024-02-16:
I have some experience with removing `dyn` (from https://github.com/tock/tock/pull/3431), so it might be interesting to try removing `dyn` from other capsules. Do you have any starting points from where I can start?