tock/tock #3879: RFC: Expand Alarm Range

Samir-Rashid, 2024-02-24:
# Issue
Tock does not support setting alarms more than 2^32 ticks in the future. 

The [syscall](https://github.com/tock/libtock-c/blob/master/libtock/alarm.h#L51-L64) offered by `libtock-c` operates in ticks.
```c
/** \brief Create a new alarm to fire at a particular clock value.
 *
 * The `alarm` parameter is allocated by the caller and must live as long as
 * the alarm is outstanding.
 *
 * \param reference: the reference time from which the alarm is being set
 * \param dt: the time after reference that the alarm should fire
 * \param callback a callback to be invoked when the alarm expires.
 * \param userdata passed to the callback.
 * \param a pointer to a new alarm_t to be used by the implementation to keep
 *        track of the alarm.
 * \return An error code. Either RETURNCODE_SUCCESS or RETURNCODE_FAIL.
 */
int alarm_at(uint32_t reference, uint32_t dt, subscribe_upcall, void*, alarm_t *alarm);
```
<br>
@tyler-potyondy, @atar13 and I are porting Openthread to Tock. Openthread requires access to a 2^32 ms timer (~= 50 days). 

Currently, the Tock kernel keeps track of `Time` in ticks. This is problematic because, for a nrf53840dk, `2^32 ticks / frequency` is only about 2 days. Adding an interface to set alarms in seconds or milliseconds would be useful.

# Potential Options
Long running timer support could be added in the kernel or through the `libtock` interface. This can be done by storing a `counter: u32` which counts the number of overflows until the alarm should get scheduled.  

## Kernel
Adding a milliseconds interface to the kernel adds more complexity. Using milliseconds would be in violation of [TRD105-time](https://github.com/tock/tock/blob/master/doc/reference/trd105-time.md#4-alarm-and-alarmclient-traits), which only specifies ticks for everything. However, adding another option is still backwards compatible.

## `libtock-[c,rs]`
`libtock` could offer another [`alarm_at`](https://github.com/tock/libtock-c/blob/master/libtock/alarm.h#L51-L64) that takes human understandable units as opposed to ticks. This would be difficult as libtock does not have access to the `overflow_client` and cannot easily do the housekeeping of incrementing an overflow count.
<br>
CC: @alevy @phil-levis Relevant TRD contributors

tyler-potyondy, 2024-02-26:
This is somewhat trivial to implement in userspace as you suggested @alevy and will not in any way impede the openthread port. Perhaps the broader question is if Tock should support providing an 8byte tick expiration instead of a 4byte tick value (i.e. the set alarm syscall uses two arguments to provide high / low bits). From this, the `alarm` capsule would then internally handle the overflow. The drawback to this is added complexity within the capsule. However, this would simplify the logic libtock implements and also avoid needing to reimplement this in each libtock variant (c / rs / future).  

The answer to all of this may perhaps be that "wrapping alarms" are something userspace should handle. If so the userspace implementation can easily be completed for libtock-c. This is a broader design philosophy of where the implementation complexity should reside.

For sake of discussion and context: for the default nrf52840 tock configuration at 32KHz, the greatest possible value for the syscall alarm is 37 hours.

bradjc, 2024-02-26:
`timer_in(ms)` doesn't work?

```c

/** \brief Create a new alarm to fire in `ms` milliseconds.
 *
 * \param ms the number of milliseconds to fire the alarm after.
 * \param callback a callback to be invoked when the alarm expires.
 * \param userdata passed to the callback.
 * \param A handle to the alarm that was created.
 * \return An error code. Either RETURNCODE_SUCCESS or RETURNCODE_FAIL.
 */
int timer_in(uint32_t ms, subscribe_upcall, void*, tock_timer_t* timer);
```

tyler-potyondy, 2024-02-27:
`timer_in` works in the sense of providing the time in milliseconds, but is flawed in that it is just a wrapper around `alarm_at` which can only accept a max value of  $2^32$ ticks

lschuermann, 2024-02-27:
Given the large range of underlying frequencies we support for these timers (especially the nRF example above is useful), I think that we should then fix `timer_in` to repeatedly wait for the max ticks that the alarm syscall driver supports, if the wait time were to overflow.

I agree with most comments above in that I don't believe the syscall driver needs to be adjusted, because we do clearly communicate those limits (through the number of bits available to the ticks argument).

alevy, 2024-02-28:
For completeness, I think questioning whether the system call interface should support this is reasonable, in general. There are multiple competing tradeoffs to consider:

1. It is possible on some hardware to do this in the kernel in a more power efficient manner. Basically if you can setup a counter to increment when the timer wraps around, and set an interrupt to fire when the counter reaches a certain value, you can do this without waking the CPU up until very close to the deadline. I _believe_ this is possible on the NRF

2. However, that would add a fair bit of code size and complexity regardless of whether apps actually used that system call ever

3. And is probably not supportable on all platforms, so would introduce even more code size and complexity in chips that _didn't_ support it to simulate it.

4. On the other other hand, doing this purely in userspace, in practice, adds some code size complexity for each application that uses the timer (since it'll be in the library code that implements the timer) regardless of whether it's useful to that app.

It's not immediately obvious how these tradeoffs play-out generally _but_ I think an intuition is that such a long timer is probably rare enough that it's OK to force that application to figure out how to handle this on its own _or_ the added code to support this in the libtock-c timer library is minimal enough to just do it there.

alevy, 2024-02-28:
I think this is a very valuable thing to have raised and it's good to have it for posterity. @Samir-Rashid can you close the issue if you think the responses resolve this uncontroversially for you? (DO NOT CLOSE IT IF YOU ARE NOT CONVINCED)