tock/tock #3922: Writing the Tock OS kernel to the stm32f303discovery development board failed.

LoveRikka, 2024-03-15:
As indicated in the title, I encountered the following error when using the make flash command:
> loverikka@localhost:~/rust_embeded/tock/boards/stm32f3discovery$ make flash
    Finished release [optimized + debuginfo] target(s) in 0.03s
   text    data     bss     dec     hex filename
 100354      32   18220  118606   1cf4e /home/loverikka/rust_embeded/tock/target/thumbv7em-none-eabi/release/stm32f3discovery
openocd -f openocd.cfg -c "init; reset halt; flash write_image erase /home/loverikka/rust_embeded/tock/target/thumbv7em-none-eabi/release/stm32f3discovery.elf; verify_image /home/loverikka/rust_embeded/tock/target/thumbv7em-none-eabi/release/stm32f3discovery.elf; reset; shutdown"
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1000 kHz
Info : STLINK V2J16S0 (API v2) VID:PID 0483:3748
Info : Target voltage: 2.913163
Info : [stm32f3x.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32f3x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f3x.cpu on 3333
Info : Listening on port 3333 for gdb connections
[stm32f3x.cpu] halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
Info : device id = 0x10036422
Info : flash size = 256 KiB
Info : Flash write discontinued at 0x08018800, next section at 0x0801ffe0
Error: error writing to flash at address 0x08000000 at offset 0x00000000
auto erase enabled
make: *** [Makefile:25: flash] Error 1

The error mainly manifests in the following two lines:

Info: Flash write discontinued at 0x08018800, next section at 0x0801ffe0
Error: error writing to flash at address 0x08000000 at offset 0x00000000.

I can confirm that it's not an issue with ST-Link because when I tried the Rust HAL tutorial, I was able to successfully use OpenOCD to write.

My openocd.cfg file is configured as follows：

```

# Licensed under the Apache License, Version 2.0 or the MIT License.
# SPDX-License-Identifier: Apache-2.0 OR MIT
# Copyright Tock Contributors 2023.

#interface
interface hla
hla_layout stlink
hla_device_desc "ST-LINK/V2"
hla_vid_pid 0x0483 0x3748

# The chip has 48KB sram
set WORKAREASIZE 0xC000

# Revision C (newer revision)
#source [find interface/stlink-v2-1.cfg]

# Revision A and B (older revisions)
#source [find interface/stlink-v2.cfg]

source [find target/stm32f3x.cfg]

```

Thanks to everyone in the community

bradjc, 2024-03-15:
I've had trouble using OpenOCD with STM32 boards as well.

Is this https://github.com/rust-embedded/cortex-m-quickstart/blob/master/openocd.cfg the openocd cfg file that worked for you? Could you try replacing the openocd cfg file in tock with the what you were able to use?

LoveRikka, 2024-03-16:
> I've had trouble using OpenOCD with STM32 boards as well.
> 
> Is this https://github.com/rust-embedded/cortex-m-quickstart/blob/master/openocd.cfg the openocd cfg file that worked for you? Could you try replacing the openocd cfg file in tock with the what you were able to use?

I'm sorry, the cfg file you shared isn't useful for me. However, I found its openocd.gdb file in the Rust HAL file, you can try converting it into an openocd.cfg file that suits your needs. The configuration is as follows:
```
# Connect to gdb remote server
target remote :3333

# Load will flash the code
load

# Enable demangling asm names on disassembly
set print asm-demangle on

# Enable pretty printing
set print pretty on

# Disable style sources as the default colors can be hard to read
set style sources off

# Initialize monitoring so iprintln! macro output
# is sent from the itm port to itm.txt
monitor tpiu config internal itm.txt uart off 8000000

# Turn on the itm port
monitor itm port 0 on

# Set a breakpoint at main, aka entry
break main

# Set a breakpoint at DefaultHandler
break DefaultHandler

# Set a breakpoint at HardFault
break HardFault

# Continue running until we hit the main breakpoint
continue

# Step from the trampoline code in entry into main
step

```
The function of the openocd.gdb file seems to be different from that of the openocd.cfg file. I'm not sure if this will work.

bradjc, 2024-03-18:
I don't think we need the GDB file for flashing. Do you happen to know which openocd config file did allow you to flash the stm32?

It would be great to fix this for others, but I don't know what a working openocd.cfg file should look like.

bradjc, 2024-03-22:
Now that #3926 has been merged, can you try again? That might fix the openocd issues.

LoveRikka, 2024-03-29:
> Now that #3926 has been merged, can you try again? That might fix the openocd issues.

Sorry, I'm an amateur developer, so my responses may be delayed. Thank you. After updating the local repository, I was able to successfully write the Tock image. Despite encountering some warnings, the demo ran normally after the development board restarted. Additionally, although the cfg file has been removed, I found the cfg file in the official Rust example to be very rudimentary, so it is not suitable for projects like Tock OS.

bradjc, 2024-04-05:
Please re-open if still an issue, but using the built-in openocd rules should address this.

closed 2024-04-05 by bradjc