tock/tock #3923: 6LoWPAN Indexing Bug

tyler-potyondy, 2024-03-15:
In testing the OpenThread stack, I have encountered a bug that occurs seemingly at random when handling 6LoWPAN packets. By default, we register the 6LoWPAN `RXClient` for the MAC-layer radio (IEEE802.15.4). As such, received 15.4 packets are passed to the OpenThread userprocess, but also upwards to the 6LoWPAN `RXClient`. On occasion, the 6LoWPAN capsule panics while processing the 6LoWPAN packet due to an indexing error:

```
panicked at capsules/extra/src/net/sixlowpan/sixlowpan_state.rs:744:19:
range end index 2120 out of range for slice of length 1280
```

I have observed this bug when handling approximately 5-10% of 6LoWPAN packets. This will need to be resolved in order to provide support for OpenThread. 

### Input Requested

As a side note, this bug seems to highlight another discussion for our 15.4 stack design. By having a "promiscuous receive" that passes upwards all packets and by default registers clients, we increase the overhead for handling received packets and also expose a larger amount of potential logic bugs unnecessarily. It may be fruitful to rethink when and what we automatically subscribe as RXClients.

bradjc, 2024-03-15:
Could it be corrupted packets that cause the parsing to want to extend beyond the buffer?

tyler-potyondy, 2024-03-15:
Both cases seem very likely as the culprit. The 6LoWPAN implementation of `receive` for the `RXClient` is likely missing an error check to catch such cases. I am not super familiar with the 6LoWPAN capsule, so I will need to dig a bit further into the details on what's happening.

tyler-potyondy, 2024-03-28:
After looking into this further, it appears the cause of this bug was introduced in #3919. This PR allowed for encrypted packets to be passed to userspace without being decrypted (needed for openthread). However, this had the unintended consequence of also passing encrypted packets to all other `RxClients` (current 15.4 design is for framer to pass received packets to all registered receiving clients). 

In the case of 6LoWPAN, the 6LoWPAN capsule logic assumes that it operates upon a packet that is either not secured with link layer security, or one which has been decrypted. As such, the `datagram_length` read from the 6LoWPAN header was encrypted and would sometimes be greater than the 6LoWPAN packet length (leading to the slice index panic).  To resolve this, I propose adding an encrypted flag to the `device::RxClient` `receive` method:

```rust
    fn receive<'a>(
        &self,
        buf: &'a [u8],
        header: Header<'a>,
        data_offset: usize,
        data_len: usize,
        encrypted: bool,
    );
```

In the current design, `RxClients` receive all 15.4 traffic and are expected to filter packets accordingly. Having the encrypted flag would allow for raw receiving while also simultaneously enabling each respective `RxClient` to determine a policy for handling packets that remain encrypted.

closed 2024-04-29 by tyler-potyondy