tock/tock #3938: Alarm: Change prescaler for NRF 5x boards

Samir-Rashid, 2024-03-27:
@tyler-potyondy, @atar13 and I are porting Openthread to Tock.

We are running into an issue (https://github.com/tock/libtock-c/issues/381) where the board's time counter is resetting every 7 minutes (512 seconds). We are using the [`alarm_internal_read` function of libtock-c](https://github.com/tock/libtock-c/blob/e5c540b28f672869ccaf08c0738f355ca53c780b/libtock/internal/alarm_internal.c#L24-L27) which enters [this system call](https://github.com/tock/tock/blob/22ddb0b9e5c40dc731d4e6055b61e2feb2bf5533/capsules/core/src/alarm.rs#L204-L207). This `now` value comes from [`self.alarm.now()`](https://github.com/tock/tock/blob/22ddb0b9e5c40dc731d4e6055b61e2feb2bf5533/capsules/core/src/alarm.rs#L196). For our board (the nrf52840dk), it's using the RTC counter due to [this line in the board's main.rs](https://github.com/tock/tock/blob/22ddb0b9e5c40dc731d4e6055b61e2feb2bf5533/boards/nordic/nrf52840dk/src/main.rs#L488).

In the `nrf5x`'s `rtc.rs` file we found the implementation of the `now` function:
https://github.com/tock/tock/blob/22ddb0b9e5c40dc731d4e6055b61e2feb2bf5533/chips/nrf5x/src/rtc.rs#L123-L130

As you can see, the RTC can only count up to $2^{24}$ ticks. This is problematic because, for a nrf52840dk, $2^{24}$ ticks is only 512 seconds.

We noticed that the RTC driver is setting the prescaler value to`0` on the following line in `rtc.rs`:
https://github.com/tock/tock/blob/2eafd626458fa65daac7203340a15fad12de506c/chips/nrf5x/src/rtc.rs#L139

However, it makes more sense for our use case to set a larger prescaler value so that the RTC does not overflow as quickly. There are larger prescaler values we can set which will still give us millisecond level granularity. 

[image]
From: [NRF52840dk datasheet](https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.1.pdf#page=364) page 364

# Potential Options

## Change Prescaler in Kernel
All the NRF boards can use a prescaler. This will increase the time until alarm counter overflow to 131072 seconds.

$$\frac{32768Hz \mbox{ oscillator frequency}}{2^8 \mbox{ prescaler}} = 128Hz \mbox{ frequency}$$

Currently there is no prescaler (setting `0`), so the Ticks increment at 32768Hz. After using the 8-bit prescaler (setting `1`), the frequency will be 128Hz. This frequency should be enough to maintain millisecond precision and make the alarm counter overflow time more reasonable.

brghena, 2024-03-29:
A confusion on my part: using a 128 Hz clock means that you get one tick every 7.8 ms. So by definition you no longer have millisecond precision.

bradjc, 2024-04-01:
We would need to add a new Frequency type in the HIL.

closed 2024-04-19 by Samir-Rashid