tock/tock #4020: Device Passthrough

alistair23, 2024-06-06:
### Issue

The Tock syscall interface has lots of useful advantages. It allows virtulisation, platform agnostic user space applications as well being secure. But this comes at a cost. The syscall interface requires lots of buffer copying and context switching. This impacts performance and going through the Tock syscall interface is a magnitude of order slower then direct hardware access.

For most use cases this doesn't matter and the advantages outweigh the performance impacts. The general exception to this is BLE, where the syscall overhead is too large to meet the strict BLE timing.

### Possible solutions

The generally agrees solution for BLE is to run BLE in the kernel, see https://github.com/tock/tock/issues/2970.

This obviously is a large amount of work and has the massive downside of including a bunch of C code in the Tock kernel. Even if a Rust BLE solution exists, dealing with complex network traffic in the kernel exposes us to bugs, which can break the entire system as it's running in the kernel privilege.

### Proposed solution

I wanted to propose support for device passthrough. This is commonly used in Hypervisors (https://developer.ibm.com/tutorials/l-pci-passthrough for example) to assign a device to a specific guest.

In the case of Tock we would assign a specific device to a single userspace application, probably just the first.

The high level idea is that

1. A board would optionally choose to allocate a device to an application
2. A capsule would tell the application about the exposed device and the memory addresses. This application would also be used to pass interrupts to the application via the usual Tock callback system.
3. The board would allocation a special MPU/PMP region to an application to allow it direct access to the hardware. At this point the kernel can't access the hardware and neither can any other application. It's assigned to a single application
4. The kernel will still handle interrupts and pass that to the application. It would be up to the application to handle the interrupt before the kernel re-enables them.

I think this would meet BLE timing, even with alarms interrupts going through the kernel.

#### Advantages

This can be used to run a BLE stack in a less privileged app. That way the BLE stack runs outside the kernel, while still meeting timing. It also allows running existing BLE stacks that presumably have been certified

This could also be used for other hardware, but BLE is the one that jumps out.

#### Disadvantageous

- We can't virtulise the BLE stack, the application would need to use IPC to do that for us
- The BLE implementation wouldn't be generic, so each board would need it's own implementation. Hopefully we can just port NimBLE and use that without too much extra effort for each new platform
- Security (see below)

#### Security

In server class systems we use a IO-MMU to avoid security issues from the passthrough device accessing memory it shouldn't. That is a concern in Tock although the lack of DMA in most systems should help a bit. A FIFO only BLE interface shouldn't allow a malicious application to read/write memory it shouldn't

The RISC-V [IOPMP](https://github.com/riscv-non-isa/iopmp-spec) spec should help fix this for RISC-V at least in the future.

A malicious app could DOS the system by not handling interrupts correctly. We could have the kernel handle and clear interrupts (and not expose that to userspace), but that might be tricky to get right with MPU/PMP alignment. High level the kernel could try to detect a interrupt storm and then just kill the app

The board would opt into the setup, so this wouldn't impact boards that don't want to use the feature.

### Next steps

If this is something that would be accepted we could start looking at how to implement this. It mostly would be designing the companion capsule, exposing the MMIO register and then porting a BLE example

lschuermann, 2024-06-07:
tl;dr: I see why this is interesting for BLE, and I'm not strictly opposed for specific peripherals, on specific chips, but I'm wary of promoting this to be a general approach / solution beyond singular peripherals.

My 2ct:

- I think there's nothing preventing you / us from implementing this right now, apart from perhaps hardware constraints for particular targets (for instance, a device's MMIO address space would need to conform to MPU/PMP expectations).

- This seems useful for various applications that require high-speed access to peripherals, and, in particular, downstream consumers. I'd be in favor of having "a story" around how these use-cases can be implemented in Tock, perhaps by having provisions for this in Tock's `ProcessStandard` implementation.

- I don't think that we'd need to change much about Tock's core kernel design to support this:
  - We can allow additional MPU regions to be accessible by a given process by passing a slice of them into a `ProcessStandard`, or passing in a closure that is able to add additional regions for individual processes, invoked somewhere around here:
    https://github.com/tock/tock/blob/51d7daef52c81a54d15b07f6e9c5ddb74d150f77/kernel/src/process_standard.rs#L1288-L1297
  - We can route interrupts using "shim" drivers in a `chip` crate, and corresponding userspace drivers (the shim drivers can directly implement the `SyscallDriver` trait).

- I feel strongly that we should avoid making this into a _general_ and _supported_ upstream mechanism, mostly for the reasons you outlined.

  Right now, each peripheral would require careful reasoning around the risks and pitfalls involved with providing direct access to it in userspace. This goes beyond DMA and uncontrolled memory accesses: for instance, a peripheral might be able to influence other subsystems with which it shares hardware, such as I/O ports, cryptographic engines, or even just the peripheral memory bus.

  We also do not have any strong guarantees around scheduling of userspace, or a notion of critical regions. It might not, for all peripherals, be safe to switch contexts between any two read/write operations. Doing so may leave the peripheral or the entire system in an undefined or unsafe state.

  In short, whether giving userspace access to hardware peripherals directly is safe depends on the peripheral, the userspace application, the kernel's scheduling policy for that app, and other system behavior. These are far too many degrees of freedom to reason about the safety of doing so in a general manner, and as such I believe that these approaches need necessarily be bespoke, carefully thought-out, and probably even separate implementations.

Now, whether we want something like this upstream for any particular peripheral, on a particular chip, is a different question. If we can argue that, e.g., giving userspace access to the radio on the nRF5x chips is generally safe, then this seems like one reasonable way forward to get these subsystems working.

Coincidentally and tangentially, I am currently looking into getting Nordic's SoftDevice Controller implementation to work. That might be an alternative approach (if we're willing to accept linking against a binary in the kernel, or if using a system like [Encapsulated Functions](https://dl.acm.org/doi/abs/10.1145/3625275.3625397) works).

bradjc, 2024-06-10:
I agree this seems like a great experimental feature. But I don't think we can say much about upstream until we better understand what APIs are needed and how it impacts the kernel implementation.

alistair23, 2024-06-21:
PR opened: https://github.com/tock/tock/pull/4044

It really has minimal impact on the kernel. You can see the entire "expose BLE to userspace" here: https://github.com/alistair23/tock/tree/alistair/device-pass-thru

That's enough for a userspace application to implement a working BLE stack

alevy, 2024-06-21:
Broadly, I'm supportive of this as an option. I do think any process with raw MMIO register access has to be considered trusted by default---for example, the vast majority of peripherals include DMA registers of some sort which, effectively, allow reading/writing arbitrary memory on the device, thus breaking memory protection safety.

I think that's a fine trade-off in some cases, and the only practical option for big drivers like BLE---I don't have other examples, but BLE is a pretty hard one to get right and the hard part is very chip-specific (unlike, say, Thread, where the chip-specific part is easy).

Indeed, we should consider ways to highly discourage use of this feature (BLE is an exceptional case), and to make it very clear how much it affects trust in processes that can get access to hardware through it.

alistair23, 2024-06-22:
> for example, the vast majority of peripherals include DMA registers of some sort which, effectively, allow reading/writing arbitrary memory on the device, thus breaking memory protection safety.

Yeah, I agree. It's possible the kernel could disable the DMA overall, but that's probably the exception.

Although I haven't checked the spec, I suspect the RISC-V IOPMP would help a lot here though

> I think that's a fine trade-off in some cases, and the only practical option for big drivers like BLE---I don't have other examples, but BLE is a pretty hard one to get right

WiFi, like on the ESP32s is another possible one.