tock/tock #4064: Doc: -Z unstable-options --document-hidden-items causes `Kernel` struct to not be documented.

bradjc, 2024-07-05:
Currently on https://docs.tockos.org/kernel/, there is no documentation for the `Kernel` struct:

[image]

That `Kernel` type is not a link.

However, when built locally:

```
cd tock/kernel
cargo doc
open ../target/doc/kernel/index.html
```

It looks as expected:

[image]

I tracked this down to using the `-Z unstable-options --document-hidden-items` flag. If I include that when building kernel docs I get the undocumented `Kernel` struct.

lschuermann, 2024-07-05:
It's hilarious that an option to document more can cause Rustdoc to generate _less_ docs. :upside_down_face:

bradjc, 2024-07-05:
The fix might be to change to `cargo doc --document-private-items` which is perhaps what we actually wanted.

PR #4068 (2024-07-08): Use `cargo doc --document-private-items`
### Pull Request Overview

I think when we added `--document-hidden-items` we really meant to use `--document-private-items`. This change not only doesn't require an unstable flag, it also generates better docs (fixes #4064) and checks more of our documentation.

I had to make a few fixes for cargo doc to succeed on hail.

### Testing Strategy

travis

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2024-07-08 by brghena