tock/tock #4100: Should double-registering a deferred call panic?

lschuermann, 2024-07-12:
Our current codebase is not consistent in where deferred calls are registered. For example, the `DateTime` component used to register a deferred call, but that obligation should fall onto the underlying RTC instance passed into to that component.

As a result of that bug, other boards that used their chip's RTC instance, but not the `DateTime` component panicked when reaching the kernel main loop.

As noted by @inesmaria08 and @alexandruradovici, this issue was particularly hard to trace down. At its core, it seems have been able to slip through review and tests because the STM board (on which that component was originally introduced) was able to double-register a deferred call (in the chip's `DefaultPeripherals` instantiation and in the `DateTime` component) without panicking.

From #4098:

> It seems then that the STM board instantiation worked, but double-registered the deferred call? Looking at the kernel code, that looks like it just works right now. I'm wondering whether it'd make sense to panic in this case (perhaps configurable behind a flag).

_Originally posted by @lschuermann in https://github.com/tock/tock/pull/4098#discussion_r1675967213_

bradjc, 2024-07-12:
This would panic because calling the function twice is just a strange thing to do? Or is it fundamentally wrong to double register?

lschuermann, 2024-07-12:
> This would panic because calling the function twice is just a strange thing to do? Or is it fundamentally wrong to double register?

I think more the former than the latter. Calling it twice is indicative of an error, because it implies that we're inconsistent with where we register the call for a given peripheral. Technically speaking, it doesn't break anything per se though.

bradjc, 2024-07-15:
I don't support this. Imagine two components that when used separately work just fine, but panic when used together. Or worse, the two components _do_ work together but if either is used separately they fail because the deferred call isn't registered at all.

lschuermann, 2024-07-15:
> Imagine two components that when used separately work just fine, but panic when used together.

I think the fundamental question we need to answer is whether, for any given peripheral, there should be a single canonical location where deferred calls are registered. When there's two components that do that for the same peripheral, then that's not the case.

The root cause of #4098 was that we did not have such a canonical location, and as such there was a mismatch in the assumption where the call would be registered. Having a double-register panic would have informed the author of the STM platform that there is a conflicting location around these registrations.

bradjc, 2024-07-16:
> I think the fundamental question we need to answer is whether, for any given peripheral, there should be a single canonical location where deferred calls are registered.

Yes. We should make one and this problem goes away. But making it a user problem instead of an us problem is the wrong direction.

lschuermann, 2024-07-16:
> But making it a user problem instead of an us problem is the wrong direction.

I agree. How will we enforce that we stick to our rules then? A panic is one such way that makes us notice that something's wrong, but that also has direct user impact. Is there one that users won't see? Ideally mechanical and automatic?

ppannuto, 2024-07-16:
Can we limit the scope of the ability to create deferred calls? Ideally mechanistically via rust, but in the worst case even with a naive text-parsing linter?

bradjc, 2024-07-16:
I think we don't need to over engineer this. The main problem is when a deferred call is _not_ registered. If we can fix that problem, we can avoid double-registering with just code review. Basically, no one will put deferred call registers anywhere but the canonical place once there is a canonical place, and nobody thinks about deferred calls once they work.

lschuermann, 2024-11-10:
Closing this as there doesn't seem to be a want for this.

closed 2024-11-10 by lschuermann