tock/tock #4104: Defensive programming in MPU driver API

enjhnsn2, 2024-07-15:
Hello,
While looking at the cortex-M MPU drivers, I noticed some potential integer overflow errors that could cause panics and/or quietly break process memory isolation. I'm fairly confident these APIs aren't directly exposed to userspace processes, so its not an immediate issue, but considering that this is a security-critical API of a security-focused OS, it might be worthwhile being a bit paranoid.

An example: 
The MPU API is parameterized by `MIN_REGION_SIZE`:
https://github.com/tock/tock/blob/b73d01084367b23d2abed4159cd570192bf454fb/arch/cortex-m/src/mpu.rs#L373-L374
which is then used in the function `allocate_region`:
https://github.com/tock/tock/blob/b73d01084367b23d2abed4159cd570192bf454fb/arch/cortex-m/src/mpu.rs#L443
This code will crash the kernel if `MIN_REGION_SIZE` = 0.

A more interesting example:
The `CortexMRegion::new()` function computes the size of the region by finding the log base two from the provided `region_size` and subtracting 1: https://github.com/tock/tock/blob/b73d01084367b23d2abed4159cd570192bf454fb/arch/cortex-m/src/mpu.rs#L308
However, `log_base_two` is implemented such that `log_base_two(0)` = 0. This means that if you pass in a `region_size` of 0, there will be an integer underflow, resulting in a size of `u32::MAX`, which when written to the 5-bit size field on the MPU will be a size of `0b11111`, i.e., allowing access of the entire address space. 

It probably makes sense to add either some comments describing the safety conditions for these APIs, or some checks for well-formed inputs. If it makes sense to add either some comments or checks, I'd be happy to file a pr for it.

lschuermann, 2024-07-16:
Hi @enjhnsn2,

All of the above are good points. In general, the MPU (and analog mechanisms across other architectures) is integral to upholding [Tock's threat model](https://book.tockos.org/doc/threat_model/threat_model), and as such a _correct_ implementation of its drivers are crucial. In theory, this should mean that the MPU code would be particularly clear, robust, well-tested, or even formally modeled and verified---in practice, it's robustness, tests and clarity around safety invariants may be lacking. This is highlighted by issues such as #2076, #3544, #3541, #2947, #2173, #2435, ...

I think it's important to realize that these subsystems have grown organically, and it's difficult to keep track of the various (evolving) requirements and changing kernel infrastructure while not compromising clarity, or in some cases, correctness. In particular, the Cortex-M MPU implementation has fared fairly well in this regard, with very few actual bugs manifesting over the past few years. On the RISC-V front we had a more significant amount of bugs and implementation quirks; as a direct result of that, I re-designed that entire subsystem from scratch in #3597, taking into account lessons learned from the previous implementation.

I don't know whether the ARM MPU implementation is in a state that would necessitate similarly drastic action. However, we would certainly appreciate someone thinking through the various edge cases in question, writing test cases, and/or developing a formal model of it! Here's my 2ct on that issue:

- Adding unit tests would be great. You may want to take inspiration from some of the efforts I started on the RISC-V front for this:
  https://github.com/tock/tock/blob/b73d01084367b23d2abed4159cd570192bf454fb/arch/rv32i/src/pmp.rs#L876-L1121 

  I do believe that this should not require quite as much mocking of types on ARM as it did on RISC-V: because the ARM MPU is configured through a set of MMIO registers, you can allocate a region of memory with identical size and alignment to this type over initialized memory, you can "invent" (cast) one to this type and use it in a mocked `StaticRef<` as part of the larger `MPU` struct to mock interactions with hardware. For tests that touch only on the `MpuRegion` associated type, you should not be needing to mock hardware at all.

  Maybe there is even a way to write shared test cases that work on the abstract `mpu::MPU` trait, and can then be called from both the RISC-V and ARM tests to ensure that neither implementation admits illegal, and rejects legal cases?

- Carefully and formally stating invariants throughout the code can also ease reasoning through the code. I tried to establish a pattern where all invariants are checked at the top of a method, and no real changes to hardware or stored state are done until all are verified in the RISC-V PMP implementation---it could be much more formal, but maybe that could be a source of inspiration?
  https://github.com/tock/tock/blob/b73d01084367b23d2abed4159cd570192bf454fb/arch/rv32i/src/pmp.rs#L738-L832

- Cross-checking the ARM MPU and RISC-V PMP implementations may reveal bugs, or at the very least a divergence in how the `mpu::MPU` trait constraints are understood. For example, here's an interesting test case where it's not immediately clear whether the `mpu::MPU` spec admits or forbids it:
  https://github.com/tock/tock/blob/b73d01084367b23d2abed4159cd570192bf454fb/arch/rv32i/src/pmp.rs#L1081-L1084

This is a super interesting and valuable effort, and we'd be more than happy to help you in improving this subsystem!

lschuermann, 2024-07-16:
> However, `log_base_two` is implemented such that `log_base_two(0)` = 0. This means that if you pass in a `region_size` of 0, there will be an integer underflow, resulting in a size of `u32::MAX`, [...]

Apart from my high-level comments, this is definitely a bug and we should return immediately with an error for that parameter -- there's not a reasonable way we could support it.

enjhnsn2, 2024-07-17:
> I don't know whether the ARM MPU implementation is in a state that would necessitate similarly drastic action. However, we would certainly appreciate someone thinking through the various edge cases in question, writing test cases, and/or developing a formal model of it! 

From what I've seen of the implementation, something like a rewrite would not be a good use of time. I think formally modeling the Cortex-M MPU and doing some testing will be enough to provide strong correctness guarantees for the MPU code.  

> Cross-checking the ARM MPU and RISC-V PMP implementations may reveal bugs, or at the very least a divergence in how the mpu::MPU trait constraints are understood

Ya, I agree that differentially fuzzing the two architecture's MPU drivers could be quite valuable. Depending on time constraints, me or @Samir-Rashid (who is looking at the RiscV MPU implementation) may have the cycles to work on this at some point.

> Apart from my high-level comments, this is definitely a bug and we should return immediately with an error for that parameter -- there's not a reasonable way we could support it.

Sounds good. I can make that change, but the `CortexMRegion::new(...) -> CortexMRegion` is currently infallible, so this will require changing the type signature to return an `Option<CortexMRegion>`.

enjhnsn2, 2024-07-18:
> I assume this would panic first?

I made sure to test this, and under Tock's current debug/release profiles, integer underflows panic in debug mode, but not in release mode (the default behavior in Rust).

PR #4135 (2024-08-20): Add well-formedness check to CortexMRegion::new()
### Pull Request Overview

This pull request adds a well-formedness check during the creation of CortexMRegions in the Cortex-m mpu driver. This check fixes #4104 by ensuring that CortexMRegions have a non-zero size. It also adds a lower bound of 8 on the logical size of the CortexMRegion, as this is the spec-defined minimum size of a CortexM mpu region. 

### Testing Strategy

This pull request was tested by running CI tests on a fork.

### TODO or Help Wanted

N/A, PR should be complete.

### Documentation Updated

- [x] Updated the relevant  [... cut]
closed 2024-08-20 by lschuermann