tock/tock #4134: Tracking: CHERI

alevy, 2024-08-07:
Placeholder for now, to be filled with plan for upstreaming support for CHERI

ppannuto, 2024-08-13:
>  Replace ubiquitous usize with specific sized integers and explicit pointer type

I remember talking about this (long, long ago) on a core team call (or two) for a while, and explicitly and intentionally using `usize` (as opposed to the more common `uint32_t` paradigm in a lot of embedded C) IIRC as a future-proofing for a 64-bit platform option (including ease of semi-hosted / host-side testing as most-immediate use case). Is this going to be at odds with CHERI support?

alevy, 2024-08-13:
@ppannuto I don't think so, but there are places where we use `usize` and we actually mean `u32`, places where we say `usize` and we actually mean `*const _`, and places we use `usize` and mean something like "a user meaningful pointer that should only be messed with in controlled particular ways"