tock/tock #4141: cargo: build silently breaks on stable `cargo`

Samir-Rashid, 2024-08-16:
On master Tock: using Rust toolchain (nightly) and cargo version 1.77.1 (stable), `make prepush` fails on running `cd boards/nordic/nrf52840dk && cargo clippy -- -D warnings` with a misleading error message. #4075 (836e1d3d4625000abcb3159bc5c13095dbac128f) now requires that you build with unstable `cargo`. I am not sure the best solution to letting people use stable cargo if they want but also failing if the correct cargo is not installed.

---

For completeness, the error output:
```
$ cd tock/boards/nordic/nrf52840dk && cargo clippy -- -D warnings
    Checking kernel v0.1.0 (/home/mod/Documents/github/tock/kernel)
   Compiling nrf52840dk v0.1.0 (/home/mod/Documents/github/tock/boards/nordic/nrf52840dk)
error: failed to run custom build command for `nrf52840dk v0.1.0 (/home/mod/Documents/github/tock/boards/nordic/nrf52840dk)`

Caused by:
  process didn't exit successfully: `/home/mod/Documents/github/tock/target/debug/build/nrf52840dk-65bd920e6802f16b/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at boards/nordic/nrf52840dk/../../build.rs:47:13:
  Incorrect build configuration. Verify you have not unintentionally set the RUSTFLAGS environment variable.
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
```
The cause is the following block fails because `std::env::var("CARGO_ENCODED_RUSTFLAGS")` returns `Ok("")` instead of `Ok("--cfg\u{1f}cfg_tock_buildflagssentinel [...]`.

https://github.com/tock/tock/blob/186ae37c3415bbd0440d5382163518bc1262d5ce/boards/build.rs#L41-L52

bradjc, 2024-08-21:
Using stable cargo for boards other than hail is not supported, so this is the expected error (i.e. all of our build flags are getting ignored).

ppannuto, 2024-08-23:
I do agree that the error message is a bit opaque.

Perhaps it makes sense to add to the comment at `boards/nordic/nrf52840dk/../../build.rs:47:13:`, where the error message points to, to inform users that the current board requires unstable?

PR #4163 (2024-09-04): fmt: make `panic!` error messages greppable
Keep the error text contiguous in error messages so that they are easy to locate.

Fixes #4141 by making the dependency on unstable cargo for build explicit.

### Formatting

- [x] Ran `make prepush`.
closed 2024-09-04 by ppannuto