tock/tock #4157: Wrap shell.nix around a flake

GuillaumeLagrange, 2024-08-29:
Hello guys,

I am starting to experiment with tock on NixOS and noticed you have a `shell.nix`.

I was wondering if you would be interested in a PR to wrap this shell.nix in a flake which would

- pin nix dependencies ashes and make the whole nix shell deterministic
- allow the use of direnv's `use flake` to seamlessly jump into the dev shell without having to run `nix-develop` which makes you jump into a not really user friendly bash shell by default.

The idea would be to keep the shell.nix as a file if possible to keep the current behavior available, while allowing people who want to use the flake.

The downside is that it would add two files, a flake.nix and a flake.lock at the root by default, which is a standard nix setup, but we could easily move the files to a dedicated nix directory and let users point to the flake through their `.envrc`.

Anyways, what do you think about this? I'm not the most experienced nix user but I could propose a PR if you are interested!

lschuermann, 2024-08-29:
I'm not personally using Flakes, but I don't want to stand in the way of progress. That being said, I have a couple questions and concerns about Flakes:
- AFAIK they are still an experimental feature. Is that going to change any time soon? How useful are they as an experimental feature?
- Regardless of whether we include a Nix flake definition, I feel strongly about being backwards compatible with a legacy `shell.nix` file that I can enter using a `nix-shell`.
- Having things be reproducible is nice, but we'll then also need to update the Flake lock occasionally. We update the pinned Rust revision without requiring that people also install Nix to update lockfiles or hashes by simply parsing the `rust-toolchain.toml` in the repository. If we switched to Flakes and required regenerating lockfiles on updates, we should do so in CI and ensure that folks can simply copy a new lockfile from there into their local checkout, without requiring anybody to install Nix.
- There might be pushback against additional files in the repository root, but I don't see a way around this -- a many tools require that.

ppannuto, 2024-09-04:
> There might be pushback against additional files in the repository root, but I don't see a way around this -- a many tools require that.

Yeah, it's already frustrating that the "front page" of the [repo view] of the project has so many files that one has to scroll more than a full page to get to any meaningful content. While that is a github UX issue, it also holds true that there is merit in minimizing the noise for the "`ls` of a new project I just checked out".

If putting these two files [and now also `shell.nix`??] into a `.nix/` folder is feasible, that's a much more amenable outcome I think.

---

I don't know/understand the tool well enough to understand the actual usability experience here, but I'd also be wary about commits for "lockfile churn" — as Leon identified, we very intentionally worked to remove committed lockfiles from the repo for Cargo, and if there are commits doing nothing but 'update `flake.lock`' with a greater frequency than perhaps quarterly, that would probably also have to be a discussion.

lschuermann, 2024-11-10:
Closing without prejudice, if Flakes is where the ecosystem is moving we should try to address the concerns above and create a PR. Please re-open if someone wants to pick this back up!

closed 2024-11-10 by lschuermann