tock/tock #4203: Tock OS Crashes with userspace app installed.

Arslan8, 2024-10-16:
Hi,

I am trying out Tock OS on STm32F429i and I have a simple application that does this:

```
extern int libtock_gpio_count(int *count);
int main(void) {
  printf("Hello");
  int count;
  libtock_gpio_count(&count);
  printf("Number of Pins: %d!\n", count);

  while(1) {
  for (int i =0; i<count;i++) {
    libtock_gpio_enable_output(i);
    libtock_gpio_set(i);
  }      

    printf("Setting all High!");
    
  }
}
```

but running this application crashes the OS.  Erasing the app gives me the shell back. 

Revisions: 
Tock OS: 90f746f
LibTock-C: ff1517

lschuermann, 2024-10-16:
Hi @Arslan8, could you please provide the output that indicates that the kernel is crashing? Do you get a panic dump or other error message? If not we can try to reproduce this ourselves but it'd be easiest if we had this information. Thanks!

Arslan8, 2024-10-16:
I didn't see a panic message instead I get this:
```
tock$ ��Initialization complete. Entering main loop
tock$ �������������Initialization complete. Entering main loop
tock$ �
```

The question mark is what happens when the application is loaded. Each question mark is one more restart with the application loaded. But if I debug it this is the backtrace I see:

```
#0  stm32f4xx::rcc::Rcc::new () at chips/stm32f4xx/src/rcc.rs:744
#1  0x080010c6 in stm32f429idiscovery::io::panic_fmt (info=0x1fffca88) at boards/stm32f429idiscovery/src/io.rs:79
#2  <signal handler called>

```

Which makes me say it crashed.

bradjc, 2024-10-16:
Ok yes it definitely crashed, and it seems like there are two issues:
1. Whatever caused the panic
2. The panic handler has an issue that causes it to not print anything

I'm not familiar enough with the STM32F4 to easily tell why `Rcc::new()` in the io.rs file is causing an issue. Perhaps calling it twice somehow causes the chip to hang? I'm not sure. We will look into this.

In the meantime, you can enable strace (https://github.com/tock/tock/blob/master/kernel/src/config.rs#L44) to see what is happening while the process is being loaded and executed which might provide some insight as to why having the process loaded causes the panic in the first place.

lschuermann, 2024-11-10:
Hi @Arslan8,  apologies for the delay in getting back to this.

I just tried to reproduce this on the Nucleo F429ZI board, but was unable to trigger this error. A slightly modified version of your app (to use `\r\n` line breaks for the `printf`s) instead yields the following output:

```
Initialization complete. Entering main loop
Hello
Number of Pins: 81!
Setting all High!
Setting all High!
Setting all High!
Setting all High!
Setting all High!
Setting all High!
tock$ Setting all High!
Setting all High!
Setting all High!
[...]
```

Can you confirm that this bug still exists on the latest master revisions of both the kernel and libtock-c? If it does on your hardware, I'll be happy to provide you an interactive environment connected to my Nucleo board to reproduce the issue. I'm afraid that the panic you report, at least superficially, seems like it shouldn't correlate with the behavior of this app. So without being able to reproduce this bug, it'll be like trying to find a needle in a haystack.

I'll propose we'll leave this issue open for a couple more days. If we can't reproduce it right now, we can always reopen it later when it's confirmed.

closed 2024-11-10 by lschuermann